espressif/esp-rainmaker #267: Can't get the QR code or the PoP in the monitor.

zjfls123, 2023-06-03:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

esp-idf-v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### Development Kit.

esp-rainmaker/examples/switch

### What is the expected behavior?

After downloading the example,the monitor should display the Pop and the QR code.

### What is the actual behavior?

I can't get  the QR code or the PoP,so I can't connect my esp32 with phone.The monitor displays like this: 

I (0) cpu_start: App cpu up.
I (453) cpu_start: Pro cpu start user code
I (453) cpu_start: cpu freq: 160000000
I (453) cpu_start: Application information:
I (456) cpu_start: Project name:     switch
I (461) cpu_start: App version:      1.0
I (466) cpu_start: Compile time:     Jun  3 2023 15:44:06
I (472) cpu_start: ELF file SHA256:  24453ed05ceadc6f...
I (478) cpu_start: ESP-IDF:          v4.4.4-dirty
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FCA58C0 len 00043E50 (271 KiB): D/IRAM
I (497) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (503) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (509) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (516) spi_flash: detected chip: gd
I (520) spi_flash: flash io: dio
W (524) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (538) sleep: Configure to isolate all GPIO pins in sleep state
I (544) sleep: Enable automatic switching of GPIO sleep configuration
I (551) coexist: coexist rom version e7ae62f
I (556) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) esp_rmaker_console: Initialising UART on port 0
I (577) esp_rmaker_commands: Registering command: reboot
I (577) uart: queue free spaces: 8
I (587) esp_rmaker_commands: Registering command: up-time
I (597) esp_rmaker_commands: Registering command: mem-dump
I (597) esp_rmaker_commands: Registering command: task-dump
I (607) esp_rmaker_commands: Registering command: cpu-dump
I (617) esp_rmaker_commands: Registering command: sock-dump
I (617) esp_rmaker_commands: Registering command: heap-trace
I (627) esp_rmaker_commands: Registering command: reset-to-factory
I (637) esp_rmaker_commands: Registering command: local-time
I (637) esp_rmaker_commands: Registering command: tz-set
I (647) esp_rmaker_commands: Registering command: add-user
I (647) esp_rmaker_commands: Registering command: get-node-id
I (657) esp_rmaker_commands: Registering command: wifi-prov
I (667) esp_rmaker_commands: Registering command: cmd
I (667) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3

### Steps to reproduce.

I tried to redownload the example,and build the programme and download.However,I can't change everything.

shahpiyushv, 2023-06-06:
There must be some logs after this too, indicating the provisioning status and other messages. Can you share them?

shahpiyushv, 2023-07-12:
Closing this issue because of insufficient information.

closed 2023-07-12 by shahpiyushv