espressif/esp-rainmaker #269: Thermostat Incomplete GVA

CharlesMcBaker, 2023-06-12:
### Is your feature request related to a problem?

I'm trying to create a thermostat and linking it with GVA. 

Modes work
Set Temperature works

However indoor temperature & humidity are missing in the google home.

<html>
<body>

Thermostat | esp.device.thermostat | Name, Setpoint Temperature*, AC Mode, Power
-- | -- | --

</body>
</html>

[image]

### Describe the solution you'd like.

- Add params for Humidity
- Add params & Ambient Temperature. 
-
- Add params minThresholdCelsius
- Add params maxThresholdCelsius

Since this is server side only, only Espressif can update this. 
https://developers.home.google.com/cloud-to-cloud/traits/temperaturesetting

shahpiyushv, 2023-07-12:
@CharlesMcBaker , We will add support for this soon.

shahpiyushv, 2023-07-20:
@CharlesMcBaker New release for the skills has been pushed out. Can you try again? Probably a reboot of the device may be required.