espressif/esp-rainmaker #270: Problem with WiFi setup

TKBeckum, 2023-06-14:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP- Arduino 2.0.9

### Operating System used.

macOS

### How did you build your project?

Arduino IDE

### Development Kit.

ESP-WROOM-32

### What is the expected behavior?

I try to use the rainmaker software on a device. I used the switch example. I expected in installs correctly

### What is the actual behavior?

 It is compiling and loading correctly, producing the QR-code. When using the QR-code to setup the device, the Rainmaker app is connecting, and I chose the correct network and pwd, but at the second item (confirming network) it reacts with: WiFi-status: authentication error.

### Steps to reproduce.

1. Step compiling the switch example
2. Step uploading to device
3. Step getting QR-code
4. Step Using Rainmaker app (on IOS)
5. Step App is connecting
6. Step I choose the right Wifi network and pwd
7. Step at the second item (confirming network) it reacts with: WiFi-status: authentication error.
...

### Debug Logs.

```plain
No logs
```

### More Information.

No

shahpiyushv, 2023-06-15:
WiFi-status: authentication error is indeed received if there is Wi-Fi authentication failure. On your device serial terminal, do you see that the device eventually connected to the AP, or does it just indicate error? Does your access point have any additional settings which may disallow new devices to join directly? Can you try with a different access point, may be a simple hotspot?

TKBeckum, 2023-06-15:
Thanks for your response.
I am afraid something wrong with the device. Bought three of them together,
but none of them works. Tried to establish connection to the WiFi with the
simple ESP32 WiFi example (hard coding the WiFi credentials, but that also
didn't work (see attachment; I disguised my SSID and BSSID in the output).
What is your opinion?

TKBeckum, 2023-06-15:
An update: after erasing all flash with esptool, now rainmaker is working!
Thanks for your attention.

closed 2023-06-20 by shahpiyushv