espressif/esp-rainmaker #271: Only reset WIFI but does not power off

kongshuiJ, 2023-06-19:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I found that all demos will power off and restart when resetting WIFI. May I ask if there is any interface that can only reset WIFI but does not power off?

shahpiyushv, 2023-07-12:
@kongshuiJ , currently, this is not supported. One primary reason is that we use BLE provisioning as default and once provisioning is finished, the BLE's memory is freed up. It cannot be re-allocated in the same boot cycle. However, if it is ok to retain BLE memory throughout the lifetime of the device, there is one way which was introduced some time back in [wifi provisioning](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) in esp-idf (the same component that RainMaker uses).  Going ahead, we can try using the same logic here.

kongshuiJ, 2023-07-24:
Hi @shahpiyushv .
Thank you very much for your answer.