espressif/esp-rainmaker #272: Cannot run rainmaker example

Dadeibayo, 2023-07-08:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### Development Kit.

ESP32-S3-DevKitC-1

### What is the expected behavior?

I git rainmaker's code according to the document, use vscode to use esp-idf extension, and want to run the example named switch, but vscode says that the source file "esp_rmaker_core.h" cannot be opened. I can't open any of the components. I don't know if it's related to cmakelist. I know some basic cmake, but I still don't know how to write the cmakelist file of esp-idf. There are a number of other related path issues, I wanted to use the esp32-box example to run under ESP32-S3-DevKitC-1, which involves rainmaker, but because cmake ran into a number of path issues, it is at a loss.

### What is the actual behavior?

I git rainmaker's code according to the document, use vscode to use esp-idf extension, and want to run the example named switch, but vscode says that the source file "esp_rmaker_core.h" cannot be opened. I can't open any of the components. I don't know if it's related to cmakelist. I know some basic cmake, but I still don't know how to write the cmakelist file of esp-idf. There are a number of other related path issues, I wanted to use the esp32-box example to run under ESP32-S3-DevKitC-1, which involves rainmaker, but because cmake ran into a number of path issues, it is at a loss.

### Steps to reproduce.

1. Step 略
2. Step
3. Step
...

### Build Logs.

```plain
检测到 #include 错误。请更新 includePath。已为此翻译单元(E:\espressif\project\esp-rainmaker-master\examples\switch\main\app_driver.c)禁用波形曲线。
[{
	"resource": "/e:/espressif/project/esp-rainmaker-master/examples/switch/main/app_main.c",
	"owner": "C/C++: IntelliSense",
	"code": "1696",
	"severity": 8,
	"message": "无法打开 源 文件 \"esp_rmaker_core.h\"",
	"source": "C/C++",
	"startLineNumber": 18,
	"startColumn": 1,
	"endLineNumber": 18,
	"endColumn": 29
}]
```

shahpiyushv, 2023-07-12:
@Dadeibayo , is your vs code set up correctly? Have you tried building any esp idf example?

Dadeibayo, 2023-07-12:
> @Dadeibayo , is your vs code set up correctly? Have you tried building any esp idf example?

What settings are you referring to? After I downloaded esp-idf v5.0.2, I could compile and run the example of getstart, such as blink, and then I downloaded rainmaker, and I wanted to try to run the example in rainmaker, but found it was not working.
I am a novice, how should I port the sample watering_demo under esp-box to ESP32-S3-DevKitC-1, how should I do it. At present, I have removed the relevant files with compilation errors, such as rainmaker files, wifi files, etc., and only realized the function of detecting soil moisture.
Regarding esp32-idf, I have read the documentation, but I feel that I can't grasp the key points. Maybe the documentation is still not friendly to novices, so I still can't complete it. May I ask what knowledge I need to supplement to help me better develop esp32