espressif/esp-rainmaker #274: SPIFFS & RAINMAKER

mustafakurnaz, 2023-07-29:
### Is your feature request related to a problem?

Hi, i coded a project use SPIFFS data storage. i want to add rainmaker capability. But the rainmaker partition selection does not support SPIFFS allocation. i dont know if i can use both rainmaker and SPIFFS in same time. Can you add this capability?

### Describe the solution you'd like.

rainmaker partition can allocate some memory to SPIFFS.

### Describe alternatives you've considered.

i dont know any alternatives.

shahpiyushv, 2023-07-31:
@mustafakurnaz , the partition tables in RainMaker examples are just for reference. There is some free space at the end where you can add your spiffs partition. Just ensure that you retain all the other partitions (though their addresses can change if required).

mustafakurnaz, 2023-07-31:
thanks for reply. but i dont know how to add partition to ESP32. I'm using arduino IDE and just select the partition from list. is there any source i will learn how to do?

shahpiyushv, 2023-08-01:
You can refer the docs [here](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/tutorials/partition_table.html#using-a-custom-partition-scheme) to understand how to add a custom partition table in Arduino ESP32. You can refer to [this partition table](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/rainmaker.csv) for reference, which is used in RainMaker. You can remove the coredump partition from this (basically just delete that line) and add a similar line for spiffs partition of similar or larger size (depending on available free flash).

shahpiyushv, 2023-08-11:
@mustafakurnaz , can you clarify what "sources" you mean here and where do you get the error?

mustafakurnaz, 2023-08-15:
Hi. i’m using ARDUNO IDE and the solutions i found are based on ESP-IDF structure. But then i found another way using Arduino IDE. I created a .CSV file and declared necessary sections.

# Name,   Type, SubType, Offset,  Size, Flags
nvs,data,nvs,0x9000,0x5000,
otadata,data,ota,0xe000,0x2000,
ota_0,app,ota_0,0x10000,0xF0000,
ota_1,app,ota_1,0x100000,0xF0000,
factory,app,nvs,0x1f0000,0x6000,
spiffs,data,spiffs,0x1f6000,0x186A0,
coredump, data, coredump,  0x20E6A0, 0x10000,

But program returns with;

No core dump partition found!
E (330) esp_core_dump_flash: No corE (8) SPIFFS: mount failed, -10025
An error has occurred while mounting SPIFFS

Where is the problem. I changed the partition sizes but have same problem. Thanks.

Windows için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

Kimden: Piyush ***@***.***>
Gönderilme: 11 Ağustos 2023 Cuma 06:46
Kime: ***@***.***>
Bilgi: ***@***.***>; ***@***.***>
Konu: Re: [espressif/esp-rainmaker] SPIFFS & RAINMAKER (MEGH-4850) (Issue #274)

@mustafakurnaz<https://github.com/mustafakurnaz> , can you clarify what "sources" you mean here and where do you get the error?