espressif/esp-rainmaker #275: Handshake timeout failure while provisioning

michael-drury, 2023-08-02:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-S2

### What is the expected behavior?

I expect wifi provisioning to successfully connect my ESP32S2 to the internet consistently, without failure 

### What is the actual behavior?

While connecting using the app, my mobile is able to successfully send wifi credentials to the ESP, but then fails during the "Confirming Wi-Fi connection." stage. The mobile app reads "Wi-Fi status: authentication error". 

The ESP reports a disconnect from wifi_prov_mgr of reason 15 (WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT).

The error logs suggest that this may be due to an invalid password. This isn't the case as the password supplied to the ESP has been check, and is valid.

### Steps to reproduce.

This issue was initially encountered in custom code, using host-driven claiming, but i've been able to reproduce using the standard, non-modified esp-rainmaker "switch" example (with self claiming). 

This issue is encountered only on some, but not all home network configurations.

Currently, this error is being encountered on a BT-Router, using a dual band signal with WPA-2 security.

I have attempted to connect, directly next to the router to eliminate any RSSI issues, but this also does not appear to fix the issue.

This issue doesn't appear to be new in v5.1, as I encounter the same issue in v5.0.2.

I've also attempted to enable all network compatibility options from the menuconfig in under Wi-Fi, but this doesn't appear to have helped. 

### Debug Logs.

```plain
I (49492) esp_rmaker_user_mapping: Received request for node details
I (49492) esp_rmaker_user_mapping: Got user_id = ********, secret_key = ********
I (49502) esp_rmaker_user_mapping: User Node mapping reset detected.
I (49512) esp_rmaker_user_mapping: Sending status SUCCESS
I (49582) app_wifi: Received Wi-Fi credentials
        SSID     : ********
        Password : ********
I (53432) wifi:primary chan differ, old=1, new=1, start CSA timer
I (53432) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (54252) wifi:state: init -> auth (b0)
I (54292) wifi:state: auth -> assoc (0)
I (54322) wifi:state: assoc -> run (10)
I (63342) wifi:state: run -> init (fc0)
I (63342) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (63342) app_wifi: Disconnected. Connecting to the AP again...
E (63342) wifi_prov_mgr: STA Disconnected
E (63352) wifi_prov_mgr: Disconnect reason : 15
E (63352) wifi_prov_mgr: STA Auth Error
E (63362) app_wifi: Provisioning failed!
        Reason : Wi-Fi station authentication failed
        Please reset to factory and retry provisioning
I (66192) app_wifi: Disconnected. Connecting to the AP again...
E (66192) wifi_prov_mgr: STA Disconnected
E (66192) wifi_prov_mgr: Disconnect reason : 205
```

shahpiyushv, 2023-09-04:
Would it be possible for you to try this [esp-idf example](https://github.com/espressif/esp-idf/tree/master/examples/provisioning) and provision with the [ESP Provisioning](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html#provisioning-tools) apps so that it will help identify if the issue is relevant to IDF or RainMaker?

michael-drury, 2023-09-19:
Hi Piyush, thanks for the reply. I believe esp-now init/ deinit prior to provisioning was contributing to the issue, as a stripped back reboot into provisioning mode with the latest r-maker updates seems to have fixed the issue. Closing as no longer able to reproduce.

closed 2023-09-19 by michael-drury