espressif/esp-rainmaker #276: Local control cannot control when the router has no network.

tswen, 2023-08-08:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I think I can use rmake's local control to achieve the following two points:

1. Prioritize using local control to control devices when the router can connect to the internet.
2. Use local control to control devices even when the router cannot connect to the internet.

However, based on my testing, while the first point is mostly achievable for most control scenarios, when the router cannot connect to the internet, if the ESP32 loses MQTT connectivity, local control also becomes ineffective, preventing proper control of devices within the local LAN.

shahpiyushv, 2023-08-08:
@tswen , can you also let us know the phone OS and model?

tswen, 2023-08-09:
I'm using Xiaomi 11 with MIUI version 14.0.10.

tswen, 2023-08-09:
I have noticed that when the router is unable to access the internet, local control doesn't consistently fail. This is because when the router is unable to connect to the network, the ESP32 attempts to reconnect to MQTT every 30 seconds. However, during the period just before the reconnection, local control becomes completely ineffective. This can be observed in the following logs. I am essentially controlling the device using the app every second, but there is a 20-second period where the device does not respond at all.

```
I (922390) lightbulb: 8 bit color conversion value [r:0 g:63 b:43]
I (922400) lightbulb: hal write value [r:0 g:63 b:43], channel_mask:7 fade_ms:0
I (922410) esp_rmaker_param: Reporting params: {"Light":{"Hue":161}}
I (923390) esp_rmaker_param: Received params: {"Light":{"Hue":226}}
I (923390) app_rainmaker: Received write request via : Local
I (923400) app_rainmaker: Received value = 226 for Light - Hue
I (923400) lightbulb: set [h:226 s:100 v:25]
I (923410) lightbulb: 8 bit color conversion value [r:0 g:15 b:63]
I (923410) lightbulb: hal write value [r:0 g:15 b:63], channel_mask:7 fade_ms:0
I (923420) esp_rmaker_param: Reporting params: {"Light":{"Hue":226}}
I (924310) esp_rmaker_param: Received params: {"Light":{"Hue":287}}
I (924310) app_rainmaker: Received write request via : Local
I (924320) app_rainmaker: Received value = 287 for Light - Hue
I (924320) lightbulb: set [h:287 s:100 v:25]
I (924330) lightbulb: 8 bit color conversion value [r:48 g:0 b:63]
I (924330) lightbulb: hal write value [r:48 g:0 b:63], channel_mask:7 fade_ms:0
I (924340) esp_rmaker_param: Reporting params: {"Light":{"Hue":287}}

/*           there is a 20-second period where the device does not respond at all.                 */

E (945060) esp-tls: couldn't get hostname for :a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0
E (945060) esp-tls: Failed to open new connection
E (945060) TRANSPORT_BASE: Failed to open a new connection
E (945070) MQTT_CLIENT: Error transport connect
E (945070) esp_mqtt_glue: MQTT_EVENT_ERROR
W (945080) esp_mqtt_glue: MQTT Disconnected. Will try reconnecting in a while...
W (945090) app_rainmaker: Unhandled RainMaker Common Event: 4

```