espressif/esp-rainmaker #277: Nova Home app unable to find device.

hackintoshlover, 2023-08-10:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Using esp-launchpad i uploaded the fan example to esp32.
Rainmaker android app is able to search for the device over bluetooth and adds the device.
Nova home app is unable to find the device over bluetooth.
I have to manually scan qr code to add device to the app.
Kindly fix/suggest a solution.

hackintoshlover, 2023-08-18:
Issue still remains even after updating app to v2.0.0 released on august 16 2023 on google play store.
Any steps to solve it?

shahpiyushv, 2023-08-18:
@hackintoshlover , Nova home app looks for some additional info in the BLE advertisement data to show the devices in the scan list. We have added this in the switch and led_light examples in [this commit](https://github.com/espressif/esp-rainmaker/commit/d789c719e6cf8c54d86ea2c955b78de73e090493). However, there's a bug in the Nova Home app due to which pairing with such scanned devices fails if they use a PIN for provisioning. So, these examples will work fine with Nova Home app, only if you pass `POP_TYPE_NULL` to [`app_wifi_start()`](https://github.com/espressif/esp-rainmaker/blob/6cfffd9693a067951189eb5ca8b51f976009be1a/examples/led_light/main/app_main.c#L192C11-L192C25). We will fix this bug asap.

hackintoshlover, 2023-08-18:
Ok I'll try and get back to you.
But passing pop_type_null did not help previously.

hackintoshlover, 2023-08-18:
1) So i just removed the device from app using remove button, but the device did not enter back into provision mode.
Upon pressing reset button, the device is still connected to my wifi network but does not show in app nor it enters provision mode.
I have to manually hold down boot button to reset and bring back it in provision mode.
Ideally it should happen like once device is removed from app, it goes back into provision mode (erases store wifi credentials)
Is this a bug or am i not understanding it?

2) Is there no local control in nova home app because rainmaker app commands work instantly but nova home app commands have a bit of delay. Ideally even nova home app should also have local control option.