espressif/esp-rainmaker #278: Provisioning without "pop"

hackintoshlover, 2023-08-10:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

While provisioning over ble or wifi, rainmaker app asks for "proof of possesion key".
Is there a way to provision without proof of possesion key?
Would be really helpful.

shahpiyushv, 2023-08-11:
Even though we recommend to not use it, considering the security implications, you can pass [`POP_TYPE_NONE`](https://github.com/espressif/esp-rainmaker/blob/78cd7746d08ca4f9b5d6d82841dc8b209dc258e2/examples/common/app_wifi/app_wifi.h#L38) to [`app_wifi_start`](https://github.com/espressif/esp-rainmaker/blob/78cd7746d08ca4f9b5d6d82841dc8b209dc258e2/examples/switch/main/app_main.c#L242) for provisioning without PoP.

hackintoshlover, 2023-08-11:
"POP_TYPE_NONE" results in undeclared error during compulation in arduino ide.
It works only with vscode / esp-idf.
Anyway of doing it with arduino ide?
Also nova home app does not discover devices over ble or softap, need to manually scan qrcode whereas rainmaker app has no problem in discovering devices on any method.

shahpiyushv, 2023-08-11:
For Arduino, you can simply pass NULL as the pop to [ WiFiProv.beginProvision](https://github.com/espressif/arduino-esp32/blob/e1f14331f173a00a9062f616bc9a62c358b9076f/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino#L115) call.

If the Nova Home app issue is the same as in https://github.com/espressif/esp-rainmaker/issues/277, we will check it and answer there separately.

hackintoshlover, 2023-08-11:
Passing NULL as the pop to WiFiProv.beginProvision  call worked. Thanks.
Nova Home app issue is same as #277.
Is there a way to auto enter "pop" details over ble and wifi as it happens with scanning qrcode since it will increase the user experience and remove the hassle of finding the "pop" details in an event when "pop" is unavailable.

shahpiyushv, 2023-08-13:
The QR code has the name, PoP (Proof of Possession pin) and transport info (BLE/SoftAP) which is the used to establish a secure session with the device for provisioning and that is akin to an "Auto enter". If the PoP is not available and yet you want to proceed for "better user experience" , it defeats the purpose of having a PoP in the first place. Having no PoP is the only option then.

hackintoshlover, 2023-08-13:
Understood.
Thanks.

closed 2023-08-13 by hackintoshlover