espressif/esp-rainmaker #280: Time Series PUT from mobile app

farmersfish, 2023-08-24:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi Team,

There is a case when the node doesn't have an Internet connection, and we need to save the data log (time series) to local storage, in this case, sdcard. I want to find a way to sync data from the node to the RainMaker whenever it gets the Internet connection. Is that possible?

Can I also PUT the same data log from the custom mobile phone app to the RainMaker backend? It happens when the mobile phone and the node don't have an Internet connection when they exchange data.

shahpiyushv, 2023-08-24:
> There is a case when the node doesn't have an Internet connection, and we need to save the data log (time series) to local storage, in this case, sdcard. I want to find a way to sync data from the node to the RainMaker whenever it gets the Internet connection. Is that possible?

The SDK itself does not support backing up data in case of No internet connectivity and then trying when Internet is back. However, since we use QOS1, it improves reliability in case of flaky connection. Additionally, you can monitor the `RMAKER_MQTT_EVENT_PUBLISHED` event to get assurance that MQTT publish has succeeded. Additionally, you can make use of `RMAKER_MQTT_EVENT_CONNECTED` and `RMAKER_MQTT_EVENT_DISCONNECTED` events in your application code to decide whether to store the data somewhere (if disconnected) or send right away. Handling of events can be found in the [switch example](https://github.com/espressif/esp-rainmaker/blob/6cfffd9693a067951189eb5ca8b51f976009be1a/examples/switch/main/app_main.c#L85).

> Can I also PUT the same data log from the custom mobile phone app to the RainMaker backend? It happens when the mobile phone and the node don't have an Internet connection when they exchange data.

We do not support this. Each device has its unique id and mutually authenticated session with the cloud backend. Only the device can report its data and no other entity can report it in behalf of the device.

farmersfish, 2023-08-25:
> The SDK itself does not support backing up data in case of No internet connectivity and then trying when Internet is back. However, since we use QOS1, it improves reliability in case of flaky [...]

Thanks for a great explanation!

farmersfish, 2023-08-29:
> The SDK itself does not support backing up data in case of No internet connectivity and then trying when Internet is back. However, since we use QOS1, it improves reliability in case of flaky [...]

One more question, just in case the node wasn't able update and report param due to Internet connectivity lost, can the node update param later with specific timestamp? 
esp_rmaker_param_update_and_report function seems not taking the note's timestamp as a variable. Seems as struct esp_rmaker_param
`struct esp_rmaker_param {
    char *name;
    char *type;
    uint8_t flags;
    uint8_t prop_flags;
    char *ui_type;
    esp_rmaker_param_val_t val;
    esp_rmaker_param_bounds_t *bounds;
    esp_rmaker_param_valid_str_list_t *valid_str_list;
    struct esp_rmaker_device *parent;
    struct esp_rmaker_param * next;
};`