espressif/esp-rainmaker #283: Matter example does not work with local control

tbthaker23, 2023-10-11:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32_Dev

### What is the expected behavior?

I have tried the matter light and switch example given in Rainmaker SDK. It works fine. I have enable the local control from the menuconfig settings and when I checked for local control the matter device does not controllable from local network. 

### What is the actual behavior?

Similar the Rainmaker multi_device example, local control is not working for matter_light and matter_switch. 

### Steps to reproduce.

1. Step :Enable the local control option from menuconfig
2. Step: Disable the internet and try to access the node within local network
3. Step: Node is not accessible from local network. 
...

### Debug Logs.

```plain
E (7697) Dynamic Impl: alloc(4437 bytes) failed
E (7707) esp-tls-mbedtls: write error :-0x7F00:
E (7707) transport_base: esp_tls_conn_write error, errno=Success
E (7707) mqtt_client: Writing failed: errno=0
E (7717) esp_mqtt_glue: MQTT_EVENT_ERROR
E (7717) mqtt_client: Error to subscribe topic=node/jFioTsikMxvhwyneEz9ScB/params/remote, qos=1
E (7727) esp_rmaker_param: Failed to subscribe to node/jFioTsikMxvhwyneEz9ScB/params/remote. Error -1
E (7737) esp_rmaker_core: Aborting!!!
```

shahpiyushv, 2023-10-11:
@tbthaker23 matter itself is meant for local control and so, the esp local control used in other non Matter examples is not meant to co-exist with Matter.

tbthaker23, 2023-10-11:
Thanks for quick response. I know matter itself for local control. So Matter_Light is locally controllable by using Rainmaker mobile application? When we can expect the Matter Controller for RainMaker?

closed 2023-10-11 by tbthaker23