espressif/esp-rainmaker #284: Example Switch app on Rainmaker - ESP32 resets when switched from app, switch stays on.

phuzzyday, 2023-10-16:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

2.0.14 in Arduino

### Operating System used.

Windows

### How did you build your project?

Arduino IDE

### Development Kit.

ESP-WROOM-32

### What is the expected behavior?

When pressing the switch button on the app, I expect the power to toggle on the output pin.

### What is the actual behavior?

Please note - Tried two different boards from two manufacturers with this same ESP module on them. Same result.

The ESP32 Resets, giving the following output on the terminal..

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

### Steps to reproduce.

I believe I followed the steps on the guide, (https://rainmaker.espressif.com/docs/get-started.html), as closely as expected!

I chose the Rainmaker partition scheme, and the smallest flash size was default, and correct.
The two boards tried were labelled on the bottom as; 'NODEMCU ESP-32S', and 'ESP32 DEVKITV1'.

I hope I didn't miss something really dumb. Searching brought up no help, which is unusual, so I may have screwed up in a unique and crazy way. I hope not.

I am going to try to go back one revision on the Arduino board package to see what happens. If this changes, I will post right away.

### Debug Logs.

```plain
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
```

### More Information.

The button on the board itself DOES toggle the power on the pin! When doing so, it also communicates the correct pin state to the app, so there is a lot that IS working here!
Also, after the board resets following the app switch being pressed, it goes on to still connect and indicate pin status when the button on the board is used to turn the pin off, or back on..
 Can't be a big thing... Go easy on me, I'm a bit new.

phuzzyday, 2023-10-16:
Two things I remembered too late - I set the IDE to clear all the flash before flashing.

Also, This is the output when the device was SET UP. Seemed to be lots of errors, but it did connect, so I initially disregarded.

                                      
  █▀▀▀▀▀█ ▀▀▀█▄ ▀  █▄▄▀▀█   █▀▀▀▀▀█   
  █ ███ █  ▀▄█ █▀▀▀▄█▀ █ ██ █ ███ █   
  █ ▀▀▀ █  ▄▀█▀▄  ▀▀▄▀▀██▄  █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ █▄▀ █▄█▄▀ ▀ █ ▀ █ ▀▀▀▀▀▀▀   
  ▀▀ █▀▄▀ ▄▀▄ ▀ ▀█ ▀▀█▄█ ▄▀ ▀▄▄ ▄▄▀   
   ▀▄ ▄▀▀ ▄▀  ▀  █ █▄▄▀ ▀█ █  ▄█▄█▀   
  ▄▄ ▄▄ ▀██▀█ ▄▄█▄█ ▀ ▄▀  ▀▄▀█ ▀▄▄▀   
  ▄██ █▄▀ ██  ▄ ▄▀▄  █▄ █▄▀▄▄█  ▄     
  ▄ ▀▀▄▀▀  █ █▀███ ▄▀▄▀██ ▀ ███▄ ██   
  ▄▄█▄▄▀▀▀ ▀█▄▀▄██▀ ▄██▀ ▀▄█▀▀ ▀▄▄▀   
  ▄▀▄▄▄█▀█ █ ▀█ ▄█ ▀▀█▀▀▄█▀█ ▀▄ ▄▀    
  █ █▀▄ ▀▄▄▀▄█ ▀▀▀ ▄ ▄▄▀▀█▄ ▄█ ▀▄▄█   
  ▀▀   ▀▀▀█ ▄▄ ▄▄█▄▄▀ ▀ ▀▀█▀▀▀█▄▄▀    
  █▀▀▀▀▀█  █ ▄█ ▄██ █▀▄ ███ ▀ █ ▄     
  █ ███ █ ██▀▀██ █▄▄█▄▄▄▄█▀▀▀▀▀▄▄▀▀   
  █ ▀▀▀ █ ▄▄▀▀ █▄▀█ █▀█▀ ███▄▀█ █▄█   
  ▀▀▀▀▀▀▀ ▀▀  ▀ ▀ ▀▀▀▀  ▀▀▀▀▀▀▀       
                                      

If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"RainMaker_BLE","pop":"12345678","transport":"ble"}
E (48740) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (48741) esp-tls: create_ssl_handle failed
E (48741) esp-tls: Failed to open new connection
E (48743) TRANSPORT_BASE: Failed to open a new connection
E (48749) MQTT_CLIENT: Error transport connect
E (48752) esp_mqtt_glue: MQTT_EVENT_ERROR

phuzzyday, 2023-10-16:
I have now tested the board files 2.0.12. No difference. I thought that maybe the board files for ardjuino, which also were hanging out in there, might have caused trouble. I cleared ALL files out, only installed 2.0.14, same thing.

phuzzyday, 2023-10-16:
After reading a random post where someone suggested a power supply problem, I put an oscilloscope on the 3.3V Pin on the on board regulator, and tried the operation again. The power dipped .12V below the minimum on the datasheet, and I realized with some horror that I had been using the same cable throughout all this. I put a newer, very short, cable on, and the voltage improved, but the problem remained.  So whatever rookie mistake I am making is outside of the power supply. Eliminated.

shahpiyushv, 2023-10-18:
The boot logs indicate a watchdog reset. Can you double check your GPIO interrupt/button logic to see if it can cause such issues by hogging the CPU for long?

phuzzyday, 2023-10-20:
When I started this report, I had this fear in the back of my mind that I would turn out to be a complete moron.

Well, I AM a moron, sort of. After digging around, and looking at the link I referenced as the guide I used, and looking at my code, I realized it didn't add up, as my code did NOT match the guide, or the examples code in the Arduino library, where I actually thought I got it. I was following this guide on Circuit Digest. - https://circuitdigest.com/microcontroller-projects/esp-rainmaker-tutorial-using-esp32-arduino-ide
The code from their site is what I used. It differs from the examples code, and, as you, dear reader, suspect, using the code from the examples in Arduino has set me up with zero problems.
So I guess while I AM a moron, I am not the only one, as I guess the guide has gone out of date. I am going to try to contact the authors to get a revision.

Meantime, dear helpful and supportive dev's, I sincerely apologize for wasting your time!
I hope that this will stay up so I can show Circuit Digest the situation, but I am closing this. 

Thanks guys!

closed 2023-10-20 by phuzzyday

shahpiyushv, 2023-10-20:
@phuzzyday , good to know that your problem is resolved, but what exactly was the difference which you addressed?

phuzzyday, 2023-10-20:
Sorry, wordy message...
I was using code from a guide from circuitdigest, which must have a bug in it, as when I used the code from the examples in the Arduino Library, the problem disappeared.
My skill in understanding the difference in the code is not up to snuff yet, but I will understand it soon I hope!