espressif/esp-rainmaker #286: Text with fixed-width characters (the same number of pixels) in the phone

warmtek-cell, 2023-10-26:
### Is your feature request related to a problem?

no

### Describe the solution you'd like.

Text with fixed-width characters (the same number of pixels)
Having characters with fixed width, The firmware on the ESP32 could build strings including a large number of parameters which (by inserting the CR (ASCII 13)  in the right place into the string would be seen on the phone **in the space of a single parameter** for instance in the following way:

TEMPERATURE....................25.2......C
PRESSURE...............................1.5......Bar
RPM.........................................200......turns/sec
WEIGHT....................................57......kg

With all the data **lined up correctly**

(The dots were introduced because spaces are eliminated or shortened here too)

If you could also insert some codes in the string to change color , you could **highlight those values that are incorrect.**

Thanks for your attention

shahpiyushv, 2023-11-10:
@warmtek-cell , probably it will be better to have this logic on the app side to show the values in a line. If your requirement is explicitly for text strings like `25.2 C` and not just a float value `25.2`, then it would be better to have this logic in the your firmware application logic which sets the string values using `esp_rmaker_param_update_and_report()`.

warmtek-cell, 2023-11-13:
I really meant that the app on the phone could use fixed-width characters and that the ESP32 firmware could choose to tell the app to use these characters.
Thanks for your attention.