espressif/esp-rainmaker #290: E (57534) wifi_prov_mgr: Failed to start scan

PegasusFlyMe, 2023-12-06:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

choose BLE mode ,after connect with my phone, it can not scan WiFi,i want to know how to solve it,or what should I set in  idf.py menuconfig?

E (57534) wifi_prov_mgr: Failed to start scan

[image]

shahpiyushv, 2023-12-24:
Which board and esp-idf version are you using? It would have been better to use "Runtime bug report" template to provide clearer information.

PegasusFlyMe, 2023-12-25:
ESP32C3 
version : esp-idf-v5.1.1 
When I click the "Select Network" button on the phone, the serial port displays : WiFi_ Prov_ Mgr: Failed to start scan character.
When I directly use the binary files on the blog for flash, WiFi can connect normally.（https://blog.csdn.net/weixin_42880082/article/details/130449186 ）

novflying, 2024-03-31:
[rainmaker_idf_5.1.2_wifi_prov_mgr_Failed to start scan.txt](https://github.com/espressif/esp-rainmaker/files/14814497/rainmaker_idf_5.1.2_wifi_prov_mgr_Failed.to.start.scan.txt)
Rainmaker is mainstream, use the led_light example.
Same issue, detail log is attached. The error code returned by esp_wifi_scan_start is ESP_ERR_WIFI_NOT_STARTED .

mphe, 2024-06-19:
Can confirm this issue still exists.
Tested with the "switch" and "led_light" example project and both fail with `wifi_prov_mgr` related errors.
There are no custom configurations, just the defaults.

After scanning the QR Code, there are no Wifi networks listed and the console logs `wifi_prov_mgr: Failed to start scan` errors. Same happens when pressing "Rescan" button.
When using the "Join other network" button and entering credentials manually, `Failed to set Wi-Fi configuration` and `Failed to apply Wi-Fi Credentials` errors are logged.

Board: ESP32-S3
ESP IDF Version: 5.2.2
ESP Rainmaker Version: Current master (8da28b00e17e619d0fc7d3a2dec9ab089d65b8f7)

Also using regular build procedure:
```sh
$ cd esp-rainmaker/examples/switch/
$ idf.py set-target esp32s3
$ idf.py build
$ idf.py erase-flash
$ idf.py flash monitor
```

Log:
```text
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2de0
entry 0x403c9914
I (33) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (33) boot: compile time Jun 19 2024 11:53:13
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v0.1
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (79) boot:  1 nvs_key          NVS keys         01 04 0000f000 00001000
I (87) boot:  2 nvs              WiFi data        01 02 00010000 00006000
I (94) boot:  3 otadata          OTA data         01 00 00016000 00002000
I (102) boot:  4 phy_init         RF data          01 01 00018000 00001000
I (109) boot:  5 ota_0            OTA app          00 10 00020000 00190000
I (117) boot:  6 ota_1            OTA app          00 11 001b0000 00190000
I (124) boot:  7 fctry            WiFi data        01 02 00340000 00006000
I (132) boot: End of partition table
I (136) boot: No factory image, trying OTA 0
I (141) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=4cfd0h (315344) map
I (224) esp_image: segment 1: paddr=0006cff8 vaddr=3fc9e100 size=03020h ( 12320) load
I (228) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=fe000h (1040384) map
I (477) esp_image: segment 3: paddr=0016e028 vaddr=3fca1120 size=02054h (  8276) load
I (480) esp_image: segment 4: paddr=00170084 vaddr=40374000 size=1a06ch (106604) load
I (523) boot: Loaded app from partition at offset 0x20000
I (549) boot: Set actual ota_seq=1 in otadata[0]
I (549) boot: Disabling RNG early entropy source...
I (560) cpu_start: Multicore app
I (569) cpu_start: Pro cpu start user code
I (569) cpu_start: cpu freq: 160000000 Hz
I (569) cpu_start: Application information:
I (572) cpu_start: Project name:     switch
I (577) cpu_start: App version:      1.0
I (582) cpu_start: Compile time:     Jun 19 2024 11:53:15
I (588) cpu_start: ELF file SHA256:  19961a3bb...
[... 87 lines cut ...]
I (2263) NimBLE:

I (2273) wifi_prov_mgr: Provisioning started with service name : PROV_fc0760
I (2273) app_wifi: Provisioning started
I (2283) esp_rmaker_local: Event 1
I (2283) app_wifi: Scan this QR code from the ESP RainMaker phone app for Provisioning.

[QR code here]

I (2493) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
[URL here]
I (2503) app_main: Provisioning QR : [Json here]
I (2513) app_wifi: Provisioning Started. Name : [Name here], POP : [POP here]
I (2523) app_wifi: Provisioning will auto stop after 30 minute(s).
I (17923) protocomm_nimble: mtu update event; conn_handle=1 cid=4 mtu=256
I (19073) app_wifi: Secured session established!
E (19253) wifi_prov_mgr: Failed to start scan
E (27213) wifi_prov_mgr: Failed to start scan
I (40133) esp_rmaker_user_mapping: Received request for node details
I (40133) esp_rmaker_user_mapping: Got user_id = [user id here], secret_key = [secret key here]
I (40143) esp_rmaker_user_mapping: User Node mapping reset detected.
I (40143) esp_rmaker_user_mapping: Sending status SUCCESS
E (40493) wifi_prov_mgr: Failed to set Wi-Fi configuration
E (40493) wifi_prov_handlers: Failed to apply Wi-Fi Credentials
```

sfeutrier, 2024-06-20:
Hello! Is there any update or workaround?
I have exactly this issue with ESP-IDF 5.2.2 and ESP32C3

mphe, 2024-06-25:
It works with `5.2` and `5.2.1`, so the problem appeared somewhere in `5.2.2`.

sfeutrier, 2024-06-25:
Here is my problem: https://github.com/espressif/idf-eclipse-plugin/issues/998. The version of ESP-IDF downloaded is not the tagged v5.2.2 (in fact it is but with some files with changes). I don't know if it applies for rainmaker though

shahpiyushv, 2024-07-02:
With the led_light example on esp32c3, using idf v5.2.2 worked fine for me.  Does the [wifi_prov_mgr](https://github.com/espressif/esp-idf/tree/v5.2.2/examples/provisioning/wifi_prov_mgr) example work for you with the exact same idf and board?

mphe, 2024-07-04:
The wifi_prov_mgr example fails with the same error on 5.2.2 but works with 5.2.1.
So I guess this is an ESP-IDF bug, not a RainMaker bug.

shahpiyushv, 2024-07-05:
@mphe , can you share your sdkconfig for wifi_prov_mgr example?

mphe, 2024-07-05:
[sdkconfig_5.2.1.txt](https://github.com/user-attachments/files/16105895/sdkconfig_5.2.1.txt)
[sdkconfig_5.2.2.txt](https://github.com/user-attachments/files/16105896/sdkconfig_5.2.2.txt)

mphe, 2024-07-05:
Yes, there are some modified files.
Here is a `git status` and `git diff`: [diff.txt](https://github.com/user-attachments/files/16106863/diff.txt)

mphe, 2024-07-05:
Thanks, I will try on Monday and report back.

mphe, 2024-07-16:
Indeed it works fine with latest `master`. Thanks!

closed 2024-07-23 by shahpiyushv

robocircuits, 2024-08-27:
I have v5.3 and i have the same issue. 
E (18968) wifi_prov_mgr: Failed to start scan