espressif/esp-rainmaker #294: Device Integration into Alexa

FloBer431, 2023-12-22:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I‘m working on a system for integrating firefighters pager into smart Home Systems.

My plan is, to use the device type contact sensor, as this is able to trigger routines, like switching on lights etc. 

I was able to create a contact sensor as custom Device, wich is available over the esp Rainmaker App. But Even After connecting Alexa, the Device does Not Show in the Alexa App. 
Has somebody already done something like this and had similar Problems? 

I dont know any further, i have tried many times, but it Never worked out. 

I‘m going to share some code snips After this Text.

FloBer431, 2023-12-22:
Device Declaration:
`static Device *CONTACT_SENSOR = NULL;`
...
`CONTACT_SENSOR = new Device("Contact Sensor", NULL, NULL);
Param deviceParam("Contact Detection State", "contact-detection-state" , value(false), PROP_FLAG_READ | PROP_FLAG_WRITE);
deviceParam.addUIType(ESP_RMAKER_UI_TOGGLE);
CONTACT_SENSOR->addParam(deviceParam);`
...
`
Node my_node;
my_node = RMaker.initNode("Pager");
`
...
`CONTACT_SENSOR->addCb(write_callback);`
...
`my_node.addDevice(*CONTACT_SENSOR);`

Callback Function:
`void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
  const char *device_name = device->getDeviceName();
  Serial.println(device_name);
  const char *param_name = param->getParamName();
  if (strcmp(device_name, "Contact Sensor") == 0){
    Serial.printf("ContactSensor = %s\n", val.val.b? "true" : "false");
    if (strcmp(param_name, "Contact Detection State") == 0){
      Serial.printf("Received value = %s for %s - %s\n", val.val.b ? "true" : "false", device_name, param_name);
      contact_state = val.val.b;
      param->updateAndReport(val);
    }
  }
}`

Update:
`if (contact_state != contact_state_old) {
    contact_state_old = contact_state;
    Serial.printf("Toggle State to %s.\n", contact_state ? "true" : "false");
    CONTACT_SENSOR->updateAndReportParam("Contact Detection State", contact_state);
  }`

shahpiyushv, 2023-12-22:
@FloBer431 , please refer the rainmaker [standard device types](https://rainmaker.espressif.com/docs/standard-types#devices) and [param types ](https://rainmaker.espressif.com/docs/standard-types#parameters)and create your device accordingly so that it can show up in Alexa. Let us know if that does not work.

FloBer431, 2023-12-22:
> @FloBer431 , please refer the rainmaker [standard device types](https://rainmaker.espressif.com/docs/standard-types#devices) and [param types [...]

I tried to do that, but i didn't manage to do it. Can somebody help me adjust my code or has an example for integrating a contact sensor?

FloBer431, 2023-12-27:
@shahpiyushv I tried to work with your linked types, but I'm working with the Arduino IDE library, and the whole syntax is very different, there a also not all functions, devices and params configured yet.

shahpiyushv, 2023-12-27:
While creating a device, just add the device type as second argument. For a standard contact sensor, it is "esp.device.contact-sensor", so your code will be

```
CONTACT_SENSOR = new Device("Contact Sensor", "esp.device.contact-sensor", NULL);
CONTACT_SENSOR->addNameParam();
```

This will create contact sensor and add. name parameter to it.
To create a standard parameter, you can pass its type as below
```
Param deviceParam("Contact Detection State", "esp.param.contact-detection-state" , value(false), PROP_FLAG_READ); 
```

Note that the PRO_FLAG_WRITE has been removed, since this would not be a writable parameter. Please try these changes in your code and let us know if it works.

ChrisMichae, 2024-01-03:
I am right now not in my home, I will send my updates after a month
thank you for your service

closed 2024-01-06 by FloBer431

gbrcode, 2024-02-14:
Hello, how should I do for a thermostat, what are the parameters that must be set for Alexa to recognize

shahpiyushv, 2024-02-14:
Please check out the [supported standard types](https://rainmaker.espressif.com/docs/standard-types) for this information. To summarise

device: esp.device.thermostat
parameters: esp.param.name, esp.param.setpoint-temperature, esp.param.temperature, esp.paran.ac-mode and (optionally) esp.param.power

gbrcode, 2024-02-15:
Node my_node;
  my_node = RMaker.initNode(nodeName);

  THERMOSTAT = new Device("Thermostat", "esp.device.thermostat", &gpio_dimmer);

  THERMOSTAT->addNameParam();
  THERMOSTAT->addPowerParam(DEFAULT_POWER_MODE);
  THERMOSTAT->assignPrimaryParam(THERMOSTAT->getParamByName(ESP_RMAKER_DEF_POWER_NAME));

  Param setPointParam("Target temperature", "esp.param.setpoint-temperature", value(TEMPERATURA_AMBIENTE), PROP_FLAG_READ | PROP_FLAG_WRITE);
  setPointParam.addBounds(value(0), value(100), value(1));
  setPointParam.addUIType(ESP_RMAKER_UI_SLIDER);
  THERMOSTAT->addParam(setPointParam);

  THERMOSTAT->addCb(write_callback);
  my_node.addDevice(*THERMOSTAT);

What I don't understand is how to define the other parameters, for example the temperature, and the power doesn't appear in Alexa and in Google Home it doesn't display anything. I don't know if I make myself understood. thank you

shahpiyushv, 2024-03-13:
Have you tried adding esp.param.temperature as float and esp.paran.ac-mode as string parameter since these are also mentioned to be mandatory as per our [starndard types' docs](https://rainmaker.espressif.com/docs/standard-types)?