espressif/esp-rainmaker #295: OTA with Arduino IDE project

FloBer431, 2024-01-06:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello there,

I'm trying to do my first OTA updates. But it fails everytime I try to start it in the ESP Rainmaker App.
I realized, that I wasn't able to change the version and model in the first upload to the boards.
In the Rainmaker Dashboard i read: 
- model: arduino-lib-builder
- version: esp-idf: v4.4.6 3572900934

Now my new OTA file should be named otherwise, because from now on it should has the real name, not the standardname provided by the Arduino Lib. 
But how can I change the name and version from the beginning on in the Arduino IDE, as I am planing on flashing some more modules, and I don't want to have these errors all the time.

Thanks!

Florian

PS: Sorry for my english, as I am not a native english speaker.

vikramdattu, 2024-01-08:
Hello @FloBer431 we have an Arduino project available here: https://github.com/espressif/arduino-esp32
Provides access to multiple ESP libs including RainMaker.  Can you please try these? 

Please follow the README [here](https://github.com/espressif/arduino-esp32/tree/master/libraries/RainMaker#esp-rainmaker-library-for-arduino).

vikramdattu, 2024-01-08:
@FloBer431 since, the libs in Arduino are pre-built, unfortunately, it is not possible to have custom app description information out of the box.
I found a workaround by PlatformIO user, which patches the app description afterwards. You may find the discussion [here.](https://community.platformio.org/t/how-to-engrave-application-name-version-in-esp32-partitions-with-arduino-framework/28516). [Script.](https://github.com/ankostis/Freematics/blob/dev/firmware_v5/telelogger/patchappinfos.py)

Hope this helps.