espressif/esp-rainmaker #298: A "LoadProhibited" error occurs when uploading data using ESP_Rainmaker.

EtrusUp, 2024-01-20:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### Development Kit.

ESP32-S3-SCH-V1.4

### What is the expected behavior?

Successfully created the rainmaker entity to collect my sensor data and upload it in real time!

### What is the actual behavior?

It didn't work as expected, and whenever it ran to data upload, he had the following malfunction.

### Steps to reproduce.

First of all here's my complete code.
```c
#include <string.h>
#include <stdio.h>
#include "stdbool.h"
#include "sdkconfig.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/timers.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
#include "driver/gpio.h"
#include <nvs_flash.h>

#include "onewire_bus.h"
#include "ds18b20.h"

#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_filter.h"
#include "esp_adc/adc_continuous.h"
#include "esp_adc/adc_cali_scheme.h"

#include <esp_rmaker_core.h>
#include <esp_rmaker_standard_params.h>
#include <esp_rmaker_standard_devices.h>

#include <app_wifi.h>
#include <app_reset.h>

#define ADC_UNIT ADC_UNIT_1
#define _ADC_UNIT_STR(unit) #unit
#define ADC_UNIT_STR(unit) _ADC_UNIT_STR(unit)
#define ADC_CONV_MODE ADC_CONV_SINGLE_UNIT_1
#define ADC_ATTEN ADC_ATTEN_DB_6
#define ADC_BIT_WIDTH SOC_ADC_DIGI_MAX_BITWIDTH

#define ADC_DIGI_IIR_FILTER_COEFF ADC_DIGI_IIR_FILTER_COEFF_4

[... 558 lines cut ...]
```plain
I'm sorry I don't know how to find this log, this error code is all I can see.
/*
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4005544b  PS      : 0x00060e30  A0      : 0x8200e705  A1      : 0x3fcd6530
0x4005544b: strcmp in ROM

A2      : 0x00000000  A3      : 0x3c0f6d28  A4      : 0x3fcd6560  A5      : 0x0000000c
A6      : 0x3fcd6130  A7      : 0xb33fffff  A8      : 0x00000065  A9      : 0x000000e6
A10     : 0x00000003  A11     : 0xffffffff  A12     : 0x00000050  A13     : 0x3fcd63e0
A14     : 0x3fcd61e0  A15     : 0x00000006  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x420d1d9d  LEND    : 0x420d1db8  LCOUNT  : 0x00000000  
0x420d1d9d: __lshift at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/mprec.c:486 (discriminator 1)

0x420d1db8: __lshift at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/stdlib/mprec.c:486

Backtrace: 0x40055448:0x3fcd6530 |<-CORRUPTED
0x40055448: strcmp in ROM

ELF file SHA256: 42027b2a48f5e6de
*/
I've provided my source file and hopefully it will help assist me in solving the problem. Appreciate it!
```

shahpiyushv, 2024-01-21:
The code is quite long and it would be better if you can try to isolate the problematic piece at your end first. Moreover, instead of increasing the stack size, have you tried decreasing it, so as to ensure that other modules can have enough RAM to allocate their data?

EtrusUp, 2024-01-21:
> The code is quite long and it would be better if you can try to isolate the problematic piece at your end first. Moreover, instead of increasing the stack size, have you tried decreasing it, so as to ensure that other modules can have enough RAM to allocate their data?

Thanks a lot for receiving your reply. Firstly the stack depth of the task is from 4096~256*1024 I have tried this and unfortunately this does not work, secondly as I said above my problem occurs with this line of code task:
 ```c 
 ESP_ERROR_CHECK(rmaker_report_data(water_data));
  ```
It corresponds to the following execution function:
```c
esp_err_t rmaker_report_data(water_data_t *water_data)
{
    if (water_data == NULL)
    {
        return ESP_FAIL;
    }

    ESP_ERROR_CHECK(esp_rmaker_param_update_and_report(
        esp_rmaker_device_get_param_by_type(water_sensor_device, "温度"),
        esp_rmaker_float(water_data->temp)));

    ESP_ERROR_CHECK(esp_rmaker_param_update_and_report(
        esp_rmaker_device_get_param_by_type(water_sensor_device, "TDS值"),
        esp_rmaker_float(water_data->tds)));

    ESP_ERROR_CHECK(esp_rmaker_param_update_and_report(
        esp_rmaker_device_get_param_by_type(water_sensor_device, "浊度"),
        esp_rmaker_float(water_data->turb)));

    ESP_ERROR_CHECK(esp_rmaker_param_update_and_report(
        esp_rmaker_device_get_param_by_type(water_sensor_device, "水位"),
        esp_rmaker_float(water_data->deep)));

    /* Check the current weather text */
    if ((water_data->temp) >= 40)
    {
        ESP_ERROR_CHECK(esp_rmaker_raise_alert("水温超过40℃"));
    }

    if ((water_data->tds) >= 100)
    {
        ESP_ERROR_CHECK(esp_rmaker_raise_alert("蓄水池TDS值异常"));
    }

    if ((water_data->turb) >= 1)
    {
        ESP_ERROR_CHECK(esp_rmaker_raise_alert("蓄水池浊度超过1NTU"));
    }

    return ESP_OK;
}
``` 
Every time the program runs to this step the error I mentioned occurs, only by removing this task my program can run stably, but then it won't be able to complete the data uploading and updating requirements. Thank you so much for checking out the problem for me and I hope you can assist me in solving it. I will be grateful!

EtrusUp, 2024-01-25:
> The code is quite long and it would be better if you can try to isolate the problematic piece at your end first. Moreover, instead of increasing the stack size, have you tried decreasing it, so as to ensure that other modules can have enough RAM to allocate their data?

How can I solve this problem, please?

EtrusUp, 2024-01-29:
You didn't answer my question but closed it?

shahpiyushv, 2024-01-29:
@EtrusUp , the issue in esp-idf was closed because it was a duplicate of this issue.

Meanwhile, please replace your `esp_rmaker_device_get_param_by_type()` calls by `esp_rmaker_device_get_param_by_name()` since 温度, TDS值, etc. have been added as names and not types. Internally, a NULL check is missing while comparing the types, which is causing this error. We will fix that soon, but for your specific case, just using `esp_rmaker_device_get_param_by_name()` should be sufficient.

EtrusUp, 2024-01-29:
> @EtrusUp , the issue in esp-idf was closed because it was a duplicate of this issue.
>
> Meanwhile, please replace your `esp_rmaker_device_get_param_by_type()` calls by [...]

I am indebted to you beyond words! Thank you for taking the time to address my question, even though it seems silly now, sorry I don't understand it very well yet, I'm learning.

closed 2024-01-29 by EtrusUp