espressif/esp-rainmaker #300: [esp rainmaker component] Build error with unkown type 'time_t'

lijunru-hub, 2024-02-08:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

IDF release/v5.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32S3

### What is the expected behavior?

Build Success.

### What is the actual behavior?

[image]

### Steps to reproduce.

Pull esp_rainmaker:v1.1.0, then build.

You should add #include <time.h> in the header file where you are using time_t.

lijunru-hub, 2024-03-08:
If this issue has been resolved, please let me know.

shahpiyushv, 2024-03-08:
The issue is fixed in the SDK itself in https://github.com/espressif/esp-rainmaker/commit/7d4c1e36f9110e04c30c21a708762ed91eea99ac, but the component had not been updated. We will do it with some other change soon enough.

lijunru-hub, 2024-03-08:
Thank for this.

closed 2024-03-08 by lijunru-hub