espressif/esp-rainmaker #301: Contact sensor on Arduino IDE - Trigger Alexa routine

lukeferrero, 2024-02-20:
### Is your feature request related to a problem?

I need to trigger Alexa routines, to do this I need a contact sensors or motion sensor type device. In the examples there are only "standard device". Is it possible to obtain an example with this kind of device?
I also tried to modify the switch example but those standard types seems not available.
Thank you.

shahpiyushv, 2024-02-21:
Please check out the [custom air cooler](https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/examples/RMakerCustomAirCooler/RMakerCustomAirCooler.ino) example for reference. There you will see that custom device type my.device.air-cooler has been used while [defining a device](https://github.com/espressif/arduino-esp32/blob/1847723c6e21a9de160670d462007cadfff12c95/libraries/RainMaker/examples/RMakerCustomAirCooler/RMakerCustomAirCooler.ino#L131) . Similarly, some parameters are defined like [here](https://github.com/espressif/arduino-esp32/blob/1847723c6e21a9de160670d462007cadfff12c95/libraries/RainMaker/examples/RMakerCustomAirCooler/RMakerCustomAirCooler.ino#L150), with the types (Eg. ESP_RMAKER_PARAM_MODE) passed explicitly. Just use standard sensor device types and the corresponding param types from [this page](https://rainmaker.espressif.com/docs/standard-types), and that should be sufficient.

lukeferrero, 2024-02-21:
Thank you.
I try the example.

Luca

lukeferrero, 2024-02-21:
When i try to compile with a standard sensor type like "doorbell" i've a compilation errror, "not recognized type":

    my_switch = new Doorbell("Doorbell", &gpio_switch);

It seems very simple but it doesn't work for me...
Thanks

lukeferrero, 2024-02-21:
Another question, If I have to create a new device, how should I declare a parameter so that Alexa interprets it as a contact or a motion sensor so that I can trigger a routine?
Thanks

lukeferrero, 2024-02-23:
This is code, I just need to create a device recognized by Alexa as a button to trigger a routine

//This example demonstrates the ESP RainMaker with a custom Air Cooler device
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_doorbell false

const char *service_name = "PROV_123";
const char *pop = "abcd123";

static int gpio_doorbell = 17;
static int gpio_reset = 0;

// The framework provides some standard device types like switch, lightbulb, fan, temperature sensor.
// But, you can also define custom devices using the 'Device' base class object, as shown here
static Device *my_device = NULL;

// WARNING: sysProvEvent is called from a separate FreeRTOS task (thread)!
void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
    case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32S2
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
#else
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
        printQR(service_name, pop, "ble");
#endif
        break;
    case ARDUINO_EVENT_PROV_INIT:
        wifi_prov_mgr_disable_auto_stop(10000);
        break;
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
        wifi_prov_mgr_stop_provisioning();
        break;
    default:;
    }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();

if (strcmp(param_name, "doorbell") == 0) {
        Serial.printf("\nReceived value = %s for %s - %s\n", val.val.b ? "true" : "false", device_name, param_name);
        bool doorbell = val.val.b;
        (doorbell == false) ? digitalWrite(gpio_doorbell, LOW) : digitalWrite(gpio_doorbell, HIGH);
        param->updateAndReport(val);
    } 
}

void setup()
{
    Serial.begin(115200);
    pinMode(gpio_reset, INPUT_PULLUP);
    pinMode(gpio_doorbell, OUTPUT);
    digitalWrite(gpio_doorbell, DEFAULT_doorbell);

    Node my_node;
    my_node = RMaker.initNode("ESP RainMaker Node");
    //my_device = new Device("Doorbell", "my.device.doorbell", NULL); 
    my_device = new Device("Doorbell", NULL, NULL); 
    if (!my_device) {
        return;
    }

    Param sensor("doorbell", ESP_RMAKER_PARAM_TOGGLE, value(DEFAULT_doorbell), PROP_FLAG_READ);
    sensor.addUIType(ESP_RMAKER_UI_TOGGLE);
    my_device->addParam(sensor);

    my_device->addCb(write_callback);

    //Add custom Air Cooler device to the node
    my_node.addDevice(*my_device);

    //This is optional
    // RMaker.enableOTA(OTA_USING_TOPICS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone().
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    //Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    // RMaker.enableTZService();

    //RMaker.enableSchedule();

    //RMaker.enableScenes();

    RMaker.start();

    WiFi.onEvent(sysProvEvent);  // Will call sysProvEvent() from another thread.
#if CONFIG_IDF_TARGET_ESP32S2
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{
    if (digitalRead(gpio_reset) == LOW) { //Push button pressed

        // Key debounce handling
        delay(100);
        int startTime = millis();
        while (digitalRead(gpio_reset) == LOW) {
            delay(50);
        }
        int press_duration = millis() - startTime;

        if (press_duration > 10000) {
            // If key pressed for more than 10secs, reset all
            Serial.printf("Reset to factory.\n");
            RMakerFactoryReset(2);
        } else if (press_duration > 3000) {
            Serial.printf("Reset Wi-Fi.\n");
            // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
            RMakerWiFiReset(2);
        } 

    }
    delay(100);
}

lukeferrero, 2024-02-27:
Hello, I'd like to develop a project with Espressif but I've to solve the problem described, can you help me?
Thanks in advance,
Luca

shahpiyushv, 2024-02-27:
@lukeferrero  Please check the pointers in my [comment above](https://github.com/espressif/esp-rainmaker/issues/301#issuecomment-1956571028) once again for better understanding.

Your code has the device defined as `my_device = new Device("Doorbell", NULL, NULL);`. Instead it should be  `my_device = new Device("Doorbell", "esp.device.doorbell", NULL);`

Please refer the [standard types](https://rainmaker.espressif.com/docs/standard-types/) document once again to understand valid device types and the parameters required under them.

lukeferrero, 2024-02-28:
Thank you very much Piyush Shah.

I modified the code but Alexa still to recognize the device not as a valid trigger for routine...
I also tried to modify parameter, etc. but nothing.
I just need a working example... 

luca

shahpiyushv, 2024-03-01:
Did you also you try creating the parameter using  `Param sensor("lockstate", "esp.param.lockstate", value(0), PROP_FLAG_READ);`

lukeferrero, 2024-03-02:
It works!
Thank you