espressif/esp-rainmaker #302: [Matter_light example] I cannot connect to the esp via WiFi or BLE

dovebman, 2024-02-21:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v5.2

### Operating System used.

Linux

### How did you build your project?

None

### Development Kit.

ESP32-C6-DevKitC-1

### What is the expected behavior?

Visible QR code and Visible WiFi or BLE.

### What is the actual behavior?

idf.py menuconfig -> ESP RainMaker App Wi-Fi Provisioning -> Provisioning Transport Method (Bluetooth LE) - option not available in my menuconfig
The ESP starting, the private key is generated (for the assisted claiming) but after that there is no visible QR code or link, and I can not see neither the WiFi or BLE. Hence I can not connect the esp via the phone app.

### Steps to reproduce.

1. Set menuconfig assisted claiming
2. build
3. flash
4. monitor

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xde0
load:0x4086e610,len:0x2e00
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2 2nd stage bootloader
I (24) boot: compile time Feb 21 2024 10:53:50
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (72) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (79) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (87) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (94) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (102) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (109) boot:  7 reserved         unknown          06 06 003e0000 0001a000
I (117) boot:  8 fctry            WiFi data        01 02 003fa000 00006000
I (124) boot: End of partition table
I (129) esp_image: segment 0: paddr=00020020 vaddr=42160020 size=62350h (402256) map
[... 181 lines cut ...]
W (2196) wifi:Haven't to connect to a suitable AP now!
I (2196) chip[DL]: Done driving station state, nothing else to do...
I (2206) app_matter: Commissioning window closed
I (2206) led_driver_ws2812: led set r:0, g:0, b:0
I (2216) led_driver_ws2812: led set r:0, g:0, b:0
I (2216) led_driver_ws2812: led set r:22, g:17, b:12
E (2226) esp_rmaker_core: Failed to get device certificate.
I (2236) esp_rmaker_work_queue: Work Queue created.
I (2236) esp_claim: Initialising Assisted Claiming. This may take time.
W (2246) esp_claim: Generating the private key. This may take time.
I (6576) esp_rmaker_node: Node ID ----- 6055F9F71368
I (6576) esp_rmaker_ota_using_topics: OTA enabled with Topics
I (6576) esp_rmaker_ota: OTA state = 2
I (6576) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (6586) esp_rmaker_time_service: Time service enabled
I (6596) esp_rmaker_time: SNTP already initialized.
I (6606) app_insights: Enable CONFIG_ESP_INSIGHTS_ENABLED to get Insights.
I (6606) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x131BFC00's Attribute 0x00000001 is 6055F9F71368 **********
I (6616) app_matter: RainMaker init not done. Not processing attribute update
I (6626) esp_rmaker_core: Starting RainMaker Work Queue task
I (6636) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (6636) wifi:Haven't to connect to a suitable AP now!
I (6646) esp_claim: Waiting for assisted claim to finish.
I (6656) main_task: Returned from app_main()
```

shahpiyushv, 2024-02-21:
The regular provisioning and assisted claiming concepts do not apply to matter examples. Please refer [this README](https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md) instead.

dovebman, 2024-02-21:
Thanks.
Now when I execute these commands: 
`export ESP_SECURE_CERT_PATH=/home/doveb/esp/projects/matter_light/light/managed_components/espressif__esp_secure_cert_mgr/`
`/mfg_tool.py -v 0x131B -p 0x2 -cd $RMAKER_PATH/examples/matter/mfg/cd_131B_0002.der --csv $RMAKER_PATH/examples/matter/mfg/keys.csv --mcsv $RMAKER_PATH/examples/matter/mfg/master.csv`
I get this:
`Traceback (most recent call last):
  File "/home/doveb/esp/esp-matter/tools/mfg_tool/./mfg_tool.py", line 38, in <module>
    from esp_secure_cert.tlv_format import *
ModuleNotFoundError: No module named 'esp_secure_cert'`

wangwenxue1989, 2024-04-29:
[[this README](https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md) ](https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md)  says IDF should use V4.4.4 or V5.0.1，my question is if V5.2 can be used？