espressif/esp-rainmaker #309: Unable to provision the node once empty/wrong password string is provided

mehul-hirpara-vl, 2024-04-26:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.2.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-C3-DevKitC-02 v1.1

### What is the expected behavior?

During the provisioning stage, 
1. WiFi router/AP credentials should not be stored locally at the device level until connection is success.
2. Device should respond if it is unable to connect with WiFi Router with empty password string
3. Device should allow user to reconfigure the WiFi Router credentials details if it is incorrect.

### What is the actual behavior?

During the provisioning stage,

1. Device is storing WiFi Router/AP credentials before it's verification
2. Device is keep trying to connect with the WiFi Router/AP with empty password string
3. Once device has reported that it is unable to connect with the WiFi Router/AP with provided credentials, second time during the provisioning it is not providing list of the WiFi Routers/APs available to select from the mobile app.

### Steps to reproduce.

Use case - 1

1. During provisioning share empty password string

Use case - 2

1. Provision device with incorrect password
2. Once provisioning is failed, restart provisioning process from the mobile app (ESP RainMaker App)

### Debug Logs.

```plain
E (321499) wifi_prov_mgr: STA Disconnected
E (321499) wifi_prov_mgr: Disconnect reason : 210
E (321509) wifi:sta is connecting, return error
I (324869) app_wifi: Disconnected. Connecting to the AP again...

I (3244) app_wifi: Disconnected. Connecting to the AP again...
I (5654) app_wifi: Disconnected. Connecting to the AP again...
I (8064) app_wifi: Disconnected. Connecting to the AP again...
I (10474) app_wifi: Disconnected. Connecting to the AP again...

E (68049) wifi:sta is connecting, return error
I (71409) wifi:new:<11,2>, old:<11,0>, ap:<255,255>, sta:<11,2>, prof:1
I (71419) wifi:state: init -> auth (b0)
I (71419) wifi:state: auth -> assoc (0)
I (71419) wifi:state: assoc -> run (10)
I (73379) protocomm_nimble: mtu update event; conn_handle=1 cid=4 mtu=256
I (74409) wifi:state: run -> init (f00)
I (74419) wifi:new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:1
I (74419) app_wifi: Disconnected. Connecting to the AP again...
E (74419) wifi_prov_mgr: STA Disconnected
E (74419) wifi_prov_mgr: Disconnect reason : 15
E (74429) wifi_prov_mgr: STA Auth Error
E (74429) app_wifi: Provisioning failed!
        Reason : Wi-Fi station authentication failed
        Please reset to factory and retry provisioning
I (74449) app_wifi: Failed to connect with provisioned AP, reseting provisioned credentials
I (74479) app_wifi: Secured session established!
W (74609) wifi:sta_scan: STA is connecting, scan are not allowed!
E (74609) wifi_prov_mgr: Failed to start scan
I (77789) app_wifi: Disconnected. Connecting to the AP again...
W (1801109) app_wifi: Provisioning timed out. Please reboot device to restart provisioning.
I (1802109) NimBLE: GAP procedure initiated: stop advertising.

I (1802109) NimBLE: GAP procedure initiated: stop advertising.

I (1802109) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

E (1802139) protocomm_nimble: Error setting advertisement data; rc = 30
I (1802139) wifi_prov_mgr: Provisioning stopped
I (1802139) wifi_prov_scheme_ble: BTDM memory released
I (1802149) esp_rmaker_local: Event 6
```

shahpiyushv, 2024-07-23:
Not sure if this i still an issue (and apologies for missing this out) but you also need to reset the device to factory settings as prompted by the phone app. Only then can you restart provisioning.

CentriConnect, 2024-12-06:
@shahpiyushv what command do we need to use from the Android API to reset the device to factory settings?

I've searched far and wide, and cannot find a command to do so in the espressif android documentation.

Singhsonam123, 2024-12-10:
> Not sure if this i still an issue (and apologies for missing this out) but you also need to reset the device to factory settings as prompted by the phone app. Only then can you restart provisioning.

I didn't observe if the app is prompting to do a "factory reset" before provisioning is complete.

mehul-hirpara-vl, 2024-12-10:
> Not sure if this i still an issue (and apologies for missing this out) but you also need to reset the device to factory settings as prompted by the phone app. Only then can you restart provisioning.

I didn't observe if the app is asking to perform a factory reset before provisioning is complete.

shahpiyushv, 2025-01-01:
Meanwhile, if the question is about resetting device to factory settings due to some error in provisioning, then you need to manually press and hold the BOOT button for 3+ seconds and then release it for resetting.