espressif/esp-rainmaker #314: Allow to set a static IP address

bbinet, 2024-05-13:
### Additional context.

Do you plan to allow setting static ip address for esp-rainmaker devices anytime soon?

shahpiyushv, 2024-05-13:
ESP RainMaker uses the underlying esp-idf Wi-Fi Provisioning component. Since that component itself does not provide an option to set static IP address instead of DHCP, RainMaker cannot have it. You can file an [issue under esp-idf](https://github.com/espressif/esp-idf/issues) for this feature request.

bbinet, 2024-05-14:
Thanks for your reply.
Issue created in the esp-idf repo: https://github.com/espressif/esp-idf/issues/13791