espressif/esp-rainmaker #315: Fails to compile with

ricardoquesada, 2024-06-06:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

Using Arduino Core 3.0.1 + ESP-IDF 5.1.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

esp32

### What is the expected behavior?

I expect it to compile without errors.
Used to work with Arduino Core 2.0.17 + ESP-IDF 4.4.7

Using Arduino Core as an ESP-IDF component

### What is the actual behavior?

Compilation error:

```
In function 'esp_rmaker_schedule_parse_info_and_flags',
    inlined from 'esp_rmaker_schedule_parse_json' at /home/riq/progs/esp-idf-arduino-bluepad32-template/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:813:13,
    inlined from 'write_cb' at /home/riq/progs/esp-idf-arduino-bluepad32-template/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:951:5:
/home/riq/progs/esp-idf-arduino-bluepad32-template/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:635:17: error: 'strncpy' output truncated before terminating nul copying as many bytes from a string as its length [-Werror=stringop-truncation]
  635 |                 strncpy(*info, _info, strlen(_info));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

### Steps to reproduce.

Compile Arduino Core 3.0.1 as ESP-IDF component.

### Build Logs.

```plain
n function 'esp_rmaker_schedule_parse_info_and_flags',
    inlined from 'esp_rmaker_schedule_parse_json' at /home/riq/progs/esp-idf-arduino-bluepad32-template/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:813:13,
    inlined from 'write_cb' at /home/riq/progs/esp-idf-arduino-bluepad32-template/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:951:5:
/home/riq/progs/esp-idf-arduino-bluepad32-template/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:635:17: error: 'strncpy' output truncated before terminating nul copying as many bytes from a string as its length [-Werror=stringop-truncation]
  635 |                 strncpy(*info, _info, strlen(_info));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```
```

### More Information.

There are already two pull-request that fixes this issue:

- https://github.com/espressif/esp-rainmaker/pull/313
- https://github.com/espressif/esp-rainmaker/pull/297

Would be great if you can apply one of those.

ricardoquesada, 2024-06-06:
forgot to mention that this happens when using:

```
CONFIG_COMPILER_OPTIMIZATION_PERF=y
```

in sdkconfig.defaults

shahpiyushv, 2024-10-16:
Closing this since https://github.com/espressif/esp-rainmaker/pull/297 has been merged, which seems to resolve the problem as indicated in the issue description.

closed 2024-10-16 by shahpiyushv