espressif/esp-rainmaker #319: SSL/TLS Handshake error on esp32-wrover-e kit with rainmaker matter controller example

Shivi-6, 2024-07-05:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I was trying to create a matter controller out of my esp32-wrover-e kit with 4mb flash. First of all. the partition was 2mb each which was insufficient for a binary of 2.1mb so I crreated a single partition of 3.5 mb instead of two ota partitions. After this, the build was successful and I was able to flash the binary to my board. I was also able to generate the QR code for commissioning.
While monitoring using the "idf.py monitor" and trying to connect the controller with rainamaker app, I am getting mbedtls handshake issue. Please find the logs attached and let me know what could be the solution.
[idf_py_stdout_output.pdf](https://github.com/user-attachments/files/16104632/idf_py_stdout_output.pdf)

Shivi-6, 2024-07-05:
[idf_py_stdout_output.pdf](https://github.com/user-attachments/files/16104877/idf_py_stdout_output.pdf)

Shivi-6, 2024-07-05:
[idf_py_stdout_output.pdf](https://github.com/user-attachments/files/16104899/idf_py_stdout_output.pdf)
This is the log when the controller got connected to the google home app but after the message "Device connected" appeared and I clicked the "Done" option, it showed "something went wrong" and the device did not connect. In this case, the error "***ERROR*** A stack overflow in task Tmr Svc has been detected." appears.
I am a bit new to this so I don't know exactly where the issue lies. Please help me.

vikramdattu, 2024-07-05:
Hi @Shivi-6 this looks like a memory issue. <br>
ESP32 Wrover Kit has external RAM. Could you please enable it via menuconfig, and check if it solves the issue?

Shivi-6, 2024-07-05:
I tried enabling the option in menuconfig but the build failed. With the help of "https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage" this document, I changed some configurations and the build succeeded. Attaching the log : 
[idf_py_stdout_output_new.pdf](https://github.com/user-attachments/files/16107770/idf_py_stdout_output_new.pdf)
But now there's a different error while monitoring.

vikramdattu, 2024-07-05:
Can you share more details? Where exactly the crash is happening? You may want to enable `Invoke GDB Stub on Panic option` and check the backtrace. <br>
Do you have any changes of your own which would lead to the crash?

Shivi-6, 2024-07-08:
No I do not have any changes of my own. I was just simply flashing the matter_controller example in rainmaker sdk to my esp32 board. The only change I have done is in the partition table because the 2 partitions (each of 2 MB) were not sufficient for the binary (2.1 MB) so I created a single one.
[partitions.csv](https://github.com/user-attachments/files/16122447/partitions.csv)

Can you please guide me about the steps that I need to take to crrectly work with the controller example.

Shivi-6, 2024-07-08:
[Steps for configuration.pdf](https://github.com/user-attachments/files/16122466/Steps.for.configuration.pdf)
These are the steps I tried for flashing the controller example in ESP32

vikramdattu, 2024-07-08:
Hi @Shivi-6 please increase FreeRTOS Timer task stack size by 2K and let me know if the problem persists.
You can do this via menuconfig.

`idf.py menuconfig > Component config > FreeRTOS > Kernel >  configTIMER_TASK_STACK_DEPTH`
Do set this number to 4K.

Shivi-6, 2024-07-08:
configTIMER_TASK_STACK_DEPTH was already configured to 8192. I still tried with 4K and in both the cases, the error is still there. @vikramdattu can you please confirm if the steps mentioned in https://github.com/espressif/esp-rainmaker/issues/319#issuecomment-2212956583 are correct. I just want to make sure if I am going in the right direction

Shivi-6, 2024-07-19:
Update : @vikramdattu I changed the board and now using an ESP32 with 16 MB flash but the ssl/tls error is still there..
Please let me know what can be done

vikramdattu, 2024-07-19:
Hi @Shivi-6 the issue has nothing to do with the flash. Can you try very large `CONFIG_TIMER_TASK_STACK_DEPTH`? 16KB maybe?
The issue is clearly, the stack is not sufficient. Do you have some custom code added in the callback which might be causing the issue?

Shivi-6, 2024-07-23:
Hi @vikramdattu , after changing the hardware, the stack overflow error is not there, but I am still getting errors related to MQTT saying "Failed to Initialise MQTT". How can I solve this issue? Or can I disable MQTT and connect locally?

 Also I am trying to manually perform claiming using the rainmaker cli with "**./rainmaker.py claim --matter --mac 58BF253F05A0 /dev/ttyUSB0**" but sometimes it works and sometimes while monitoring, I get the error saying "Perform Claiming using Rainmaker cli".