espressif/esp-rainmaker #320: Unable to generate esp_secure_cert.bin

yatendra98, 2024-07-22:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I've followed these steps:
$ export ESP_SECURE_CERT_PATH=/path/to/esp_secure_cert_mgr
$ esp-matter-mfg-tool -v 0x131B -p 0x2 -cd $RMAKER_PATH/examples/matter/mfg/cd_131B_0002.der --csv $RMAKER_PATH/examples/matter/mfg/keys.csv --mcsv $RMAKER_PATH/examples/matter/mfg/master.csv

I'm able to generate the partition.bin, but not able to generate esp_secure_cert.bin file in the out folder.

sayondeep, 2024-07-22:
@yatendra98 esp_secure_cert.bin is not required for test set up. 

You can go ahead with just flashing the factory_partiton by 
`esptool.py write_flash 0x3e0000 out/131b_2/<node-id>/<node-id>-partition.bin`

yatendra98, 2024-07-22:
@sayondeep I did the same and flash the code by following next steps but couldn't able to see any advertisement packet BLE/AP in Rainmaker App. I used the BLE sniffer as well to put a filter of MAC Id but couldn't see any packets. 
Hence thought of whether there is any dependency of esp_secure_cert.bin
Attaching here the screenshot of logs.

[image]

sayondeep, 2024-07-23:
@yatendra98 
You are using matter+rainmaker example where the provisioning is taken care with commissioning.
You can follow the steps [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control) to continue with the example.
You can use esp rainmaker app for commissioning and control

shahpiyushv, 2024-10-16:
Closing this as suitable answer has been provided. Moreover, just to add, the secure cert partition is generated and flashed as part of Matter claiming steps mentioned [here](https://github.com/espressif/esp-rainmaker/tree/master/examples/matter#claiming-device-certificates)

closed 2024-10-16 by shahpiyushv