espressif/esp-rainmaker #321: Get internet data from gsm module to run rainmaker

nravin5, 2024-07-25:
### Is your feature request related to a problem?

Is it possible to run rainmaker gsm module gprs data please suggest.

shahpiyushv, 2024-07-30:
Most of the RainMaker logic is independent of the underlying transport layer. So if you have a clean integration with gsm module for any esp idf examples that use the network interface (esp_netif), using RainMaker should be possible. However, the user-node association that happens during Wi-Fi provisioning needs to be looked into.

shahpiyushv, 2024-10-16:
Closing this issue. Please file a new one if some specific info is required in this context.

closed 2024-10-16 by shahpiyushv