espressif/esp-rainmaker #323: ESP32-C6 is crashing when BlueDroid is used.

P-R-O-C-H-Y, 2024-07-30:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v5.1.3-1012-gd82f9162df

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-C6 DevKitM-1 V1.0

### What is the expected behavior?

The switch example is expected to work on C6 using BlueDroid as for all other chips BlueDroid with RainMaker works fine.
In Arduino we use BlueDroid as the BLE provider, and C6 is not working.

I was able to reproduce the issue using IDF with the RainMaker switch example, changing just few things in the SDK Config.
With the same SDK config it's working fine on ESP32-C3.
Partition table was changed to just single OTA partition, to fit the binary when BlueDroid is used.

### What is the actual behavior?

Crashes on simple_ble: Configuration mismatch. Select BLE Only or BTDM mode from menuconfig.
But there is nothing in menuconfig to fix the issue.

### Steps to reproduce.

I have created a  project on GH to easily reproduce the issue.
Just build and flash.
https://github.com/P-R-O-C-H-Y/rainmaker-bluedriod/tree/main

### Debug Logs.

```plain
I (2452) BLE_INIT: Using main XTAL as clock source
I (2462) BLE_INIT: ble controller commit:[f74e39f]
E (2462) simple_ble: Configuration mismatch. Select BLE Only or BTDM mode from menuconfig
E (2472) protocomm_ble: simple_ble_start failed w/ error code 0xffffffff
E (2482) wifi_prov_scheme_ble: Failed to start protocomm BLE service
E (2492) wifi_prov_mgr: Failed to start service
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42017bb4
0x42017bb4: app_wifi_start at /Users/prochy/esp/esp-rainmaker/examples/common/app_wifi/app_wifi.c:554 (discriminator 1)

file: "/Users/prochy/esp/esp-rainmaker/examples/common/app_wifi/app_wifi.c" line 554
func: app_wifi_start
expression: wifi_prov_mgr_start_provisioning(security, pop, service_name, service_key)

abort() was called at PC 0x4080d567 on core 0
0x4080d567: _esp_error_check_failed at /Users/prochy/esp/esp-idf/components/esp_system/esp_err.c:50

Stack dump detected
Core  0 register dump:
MEPC    : 0x40800636  RA      : 0x4080d572  SP      : 0x4082cdf0  GP      : 0x4081d400  
0x40800636: panic_abort at /Users/prochy/esp/esp-idf/components/esp_system/panic.c:466
0x4080d572: __ubsan_include at /Users/prochy/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407cbad8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x4082ce54  A0      : 0x4082ce1c  A1      : 0x4082ce52  
A2      : 0x00000000  A3      : 0x4082ce49  A4      : 0x00000001  A5      : 0x40826000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4083f0c8  S3      : 0x4083f4cc  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x4082ce1c "a") at /Users/prochy/esp/esp-idf/components/esp_system/panic.c:466
466	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x4082ce1c "a") at /Users/prochy/esp/esp-idf/components/esp_system/panic.c:466
#1  0x4080d572 in esp_system_abort (details=details@entry=0x4082ce1c "a") at /Users/prochy/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x40813eba in abort () at /Users/prochy/esp/esp-idf/components/newlib/abort.c:38
#3  0x4080d56a in _esp_error_check_failed (rc=<optimized out>, file=file@entry=0x4218ce88 "", line=line@entry=554, function=function@entry=0x4218d3e0 <__func__.1> "", expression=expression@entry=0x4218d2ec "") at /Users/prochy/esp/esp-idf/components/esp_system/esp_err.c:50
#4  0x42017bb8 in app_wifi_start (pop_type=pop_type@entry=POP_TYPE_RANDOM) at /Users/prochy/esp/esp-rainmaker/examples/common/app_wifi/app_wifi.c:554
#5  0x4200ae12 in app_main () at /Users/prochy/Gitlab/rainmaker-bluedriod/main/app_main.c:249
#6  0x42177d88 in main_task (args=<error reading variable: value has been optimized out>) at /Users/prochy/esp/esp-idf/components/freertos/app_startup.c:208
#7  0x4080fe86 in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at /Users/prochy/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:202
ELF file SHA256: cbf4015765814018

Rebooting...
```
```

vikramdattu, 2024-07-31:
Hi @P-R-O-C-H-Y thanks for taking efforts to reproduce the issue with IDF and providing the example. 

Our Bluetooth team has identified the issue and the clean fix will be provided on IDF. Meanwhile, can you please apply the below patch on IDF and check if this vents the issue for you?

```patch
diff --git a/components/protocomm/src/simple_ble/simple_ble.c b/components/protocomm/src/simple_ble/simple_ble.c
index 8a4ae0f3d9..cd6421c6e5 100644
--- a/components/protocomm/src/simple_ble/simple_ble.c
+++ b/components/protocomm/src/simple_ble/simple_ble.c
@@ -225,7 +225,7 @@ esp_err_t simple_ble_start(simple_ble_cfg_t *cfg)
#ifdef CONFIG_BTDM_CTRL_MODE_BTDM
     ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
-#elif defined CONFIG_BTDM_CTRL_MODE_BLE_ONLY || CONFIG_BT_CTRL_MODE_EFF
+#elif defined CONFIG_BTDM_CTRL_MODE_BLE_ONLY || CONFIG_BT_CTRL_MODE_EFF || CONFIG_IDF_TARGET_ESP32C6
     ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
#else
     ESP_LOGE(TAG, "Configuration mismatch. Select BLE Only or BTDM mode from menuconfig");
```

P-R-O-C-H-Y, 2024-07-31:
Hi @vikramdattu, Thank you for quick reply. Your proposed fix is working fine.
Will you also back port this fix for v5.1 release of IDF?

vikramdattu, 2024-07-31:
@P-R-O-C-H-Y yes the fix will be back ported to all the eligible IDF branches including `release/v5.1`

me-no-dev, 2024-07-31:
@P-R-O-C-H-Y please add PR with the patch to the lib-builder