espressif/esp-rainmaker #326: Homekit_switch example fails to compile

adukale, 2024-08-16:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32 Wroom Custom Board

### What is the expected behavior?

Build successful

### What is the actual behavior?

The build fails at [1368/1370] Linking CXX executable homekit_switch.elf with error message:
FAILED: homekit_switch.elf 
...
fatal_exception  -u esp_timer_init_include_func  -u uart_vfs_include_dev_init  -u include_esp_phy_override  -lphy  -lrtc  esp-idf/esp_phy/libesp_phy.a  -lphy  -lrtc  esp-idf/esp_phy/libesp_phy.a  -lphy  -lrtc  -u esp_vfs_include_console_register  -u vfs_include_syscalls_impl  -lbtdm_app && :
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj):/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:660: multiple definition of `wi_fi_scan_result__descriptor'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:1164: first defined here
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj): in function `wi_fi_scan_result__init':
/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:237: multiple definition of `wi_fi_scan_result__init'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:462: first defined here
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj): in function `wi_fi_scan_result__get_packed_size':
/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:243: multiple definition of `wi_fi_scan_result__get_packed_size'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:468: first defined here
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj): in function `wi_fi_scan_result__pack':
/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:250: multiple definition of `wi_fi_scan_result__pack'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:475: first defined here
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj): in function `wi_fi_scan_result__pack_to_buffer':
/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:257: multiple definition of `wi_fi_scan_result__pack_to_buffer'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:482: first defined here
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj): in function `wi_fi_scan_result__unpack':
/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:266: multiple definition of `wi_fi_scan_result__unpack'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:491: first defined here
/Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wifi_provisioning/libwifi_provisioning.a(wifi_scan.pb-c.c.obj): in function `wi_fi_scan_result__free_unpacked':
/Users/adwaitkale/esp/esp-idf/components/wifi_provisioning/proto-c/wifi_scan.pb-c.c:274: multiple definition of `wi_fi_scan_result__free_unpacked'; esp-idf/espressif__network_provisioning/libespressif__network_provisioning.a(network_scan.pb-c.c.obj):/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/managed_components/espressif__network_provisioning/proto-c/network_scan.pb-c.c:499: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/log/idf_py_stderr_output_14094 and /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/log/idf_py_stdout_output_14094

### Steps to reproduce.

1. idf.py build

### Build Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build
Executing "ninja all"...
[1/6] Performing build step for 'bootloader'
[1/1] cd /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/bootloader/esp-idf/esptool_py && /Users/adwaitkale/.espressif/python_env/idf5.3_py3.10_env/bin/python /Users/adwaitkale/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/bootloader/bootloader.bin
Bootloader binary size 0x68d0 bytes. 0x730 bytes (6%) free.
[2/6] No install step for 'bootloader'
[3/6] Completed 'bootloader'
[4/6] Linking CXX executable homekit_switch.elf
FAILED: homekit_switch.elf 
: && /Users/adwaitkale/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/homekit_switch.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld CMakeFiles/homekit_switch.elf.dir/project_elf_src_esp32.c.obj -o homekit_switch.elf -L/Users/adwaitkale/esp/esp-idf/components/soc/esp32/ld   -L/Users/adwaitkale/esp/esp-idf/components/esp_rom/esp32/ld   -L/Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/esp-idf/esp_system/ld   -L/Users/adwaitkale/esp/esp-idf/components/esp_phy/lib/esp32   -L/Users/adwaitkale/esp/esp-idf/components/esp_coex/lib/esp32   -L/Users/adwaitkale/esp/esp-idf/components/esp_wifi/lib/esp32   -L/Users/adwaitkale/esp/esp-idf/components/bt/controller/lib_esp32/esp32 esp-idf/xtensa/libxtensa.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp [... cut ...] if__network_provisioning/proto-c/network_scan.pb-c.c:499: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/log/idf_py_stderr_output_22014 and /Users/adwaitkale/esp-rainmaker/examples/homekit_switch/build/log/idf_py_stdout_output_22014
```

### More Information.

Tried with latest esp-idf, esp-rainmaker, esp-homekit-sdk.

adukale, 2024-08-16:
This issue is related to homekit, I guess. Other examples like switch and multi_device are getting successfully compiled.
I also tried to port homekit support into switch example but the switch example is not compiling when homekit declarations are added.

adukale, 2024-08-21:
I am able to build the homekit-switch example using this commit - https://github.com/espressif/esp-rainmaker/tree/fd78129500aa1a8b2eee9e0dcc0720d7b14cd3dc
So I think homekit_switch compilation is not working post "Replace wifi_provisioning with network_provisioning for ESP-IDF v5.1+" commit.

shahpiyushv, 2024-09-20:
Can you check if the commit https://github.com/espressif/esp-rainmaker/commit/41c7d54b1aedce68411d83899c0e092f611f0adf solves your issue?

adukale, 2024-09-26:
> Can you check if the commit [41c7d54](https://github.com/espressif/esp-rainmaker/commit/41c7d54b1aedce68411d83899c0e092f611f0adf) solves your issue?

I tried with commit [41c7d54](https://github.com/espressif/esp-rainmaker/commit/41c7d54b1aedce68411d83899c0e092f611f0adf), but same issue. Tried with latest commit also.

[image]

martinberlin, 2025-03-31:
With idf 5.5 examples are failing to build, this is what I get in the log:

```
tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj):(.data.g_wifi_osi_funcs+0x14c): undefined reference to `__wrap_esp_log_write'
/home/martin/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj):(.data.g_wifi_osi_funcs+0x150): undefined reference to `__wrap_esp_log_writev'
collect2: error: ld returned 1 exit status
```