espressif/esp-rainmaker #327: New version of Rainmaker not working

fmb787, 2024-08-20:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

iam facing when i updated the ESP Rainmaker app on IOS from V3.2.0 to  V3.2.1 , app is stuck in "confirming Node association" and receive error below and devices not added to app but the esp32 appears to be already connected to wifi , there was no changing in setting or even the code .

### i made the steps on Android app and it's working fine . ???

[image]

here is a brief of my setup 

- Iphone 11 , 17.5.1 ios 
- Arduino IDE 2.3.2
- ESP32 core 3.0.4
- ESP RainMaker App V3.2.1
- code "RMakerSwitch" sample without single change on it .  [RMakerSwitch.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino)

below is serial Monitor log 

```
21:19:02.527 -> [  6331][I][RMaker.cpp:17] event_handler(): RainMaker Initialized.
21:19:02.571 -> [  6346][I][RMaker.cpp:18] event_handler(): RainMaker Claim Started.
21:19:02.571 -> [  6353][I][WiFiProv.cpp:143] beginProvision(): Starting AP using BLE. service_name : PROV_1234, pop : abcd1234
21:19:03.021 -> 
21:19:03.021 -> Provisioning Started with name "PROV_1234" and PoP "abcd1234" on BLE
21:19:03.068 -> Scan this QR code from the ESP RainMaker phone app.

21:19:03.199 ->   █ ▀▀▀ █ ▄▄▀▀▄ ▄▀  █▀█▀▀███▄▀█ █▄█   
21:19:03.199 ->   ▀▀▀▀▀▀▀ ▀▀▀ ▀▀▀▀▀▀ ▀  ▀▀▀▀▀▀▀       

21:19:03.239 -> If QR code is not visible, copy paste the below URL in a browser.
21:19:03.239 -> https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_1234","pop":"abcd1234","transport":"ble"}
21:19:40.145 -> E (52540) esp_rmaker_core: Node connected to Wi-Fi without Assisted claiming. Cannot proceed to MQTT connection.
21:19:40.203 -> E (52540) esp_rmaker_core: Please update your phone apps and repeat Wi-Fi provisioning with BLE transport.
21:19:51.253 -> E (63647) wifi_prov_scheme_ble: bt_mem_release of BTDM failed 259
```

and after i restart the ESP32 

```
21:33:22.094 -> [   112][I][RMaker.cpp:17] event_handler(): RainMaker Initialized.
21:33:22.139 -> [   126][I][RMaker.cpp:18] event_handler(): RainMaker Claim Started.
21:33:22.139 -> [   133][I][WiFiProv.cpp:159] beginProvision(): Already Provisioned
21:33:22.139 -> [   139][I][WiFiProv.cpp:163] beginProvision(): Attempting connect to AP: xxxxxxx
21:33:22.139 -> 
21:33:22.233 -> E (220) wifi_prov_scheme_ble: bt_mem_release of BTDM failed 259
21:33:23.245 -> [  1224][E][STA.cpp:302] begin(): NETIF_STARTED_BIT not set! try to connect anyway...
```

Could someone please advise how to correct this, and whether the issue from the app or not .

.
.

- also i'll be thankful if someone could advice how to add in custom device the value ? see below 

[image]

- and one last thing i want the app to send notification when value reach a certain number , i used ( updateAndNotifyParam ) but it's not working and i got
```
Compilation error: 'class Device' has no member named 'updateAndNotifyParam'; did you mean 'updateAndReportParam'?
```

how can i do this .

thanks alot

.

fmb787, 2024-08-21:
i'll be happy for any suggestion

vikramdattu, 2024-08-22:
Hi @fmb787 couple of questions:
- Have you also tried with Android App and face same issue?
- Can you perform a factory reset (Press and hold boot button for ~10seconds) and then try provisioning again?

fmb787, 2024-08-22:
Hello @vikramdattu  thanks for your reply , 

- yes i tried with Android App and it's working fine ( i mentioned this in my discretion ) 
- yes i have perform a factory reset (10seconds) and WiFi reset (3seconds) with same issue .

fmb787, 2024-08-23:
is there any workaround way ! i can use/fix then will build app by myself .

vikramdattu, 2024-08-23:
Hi @fmb787 we are able to reproduce the issue and are looking into this.

thadius83, 2024-09-03:
> Hello @vikramdattu thanks for your reply ,
> 
> * yes i tried with Android App and it's working fine ( i mentioned this in my description)
> * yes i have perform a factory reset (10seconds) and WiFi reset (3seconds) with same issue .

Just ran into the same issue here. Same details.

AhmedShaalan, 2024-09-03:
Same Exact error for me as well. with the led_ight and switch example. I just built it, flashed it and it didn't work. 

[image]
[image]

closed 2024-09-04 by espressif-bot

shahpiyushv, 2024-09-04:
iOS app with fix for assisted claiming is now available on the [app store](https://apps.apple.com/us/app/esp-rainmaker/id1497491540) (v3.2.2)