espressif/esp-rainmaker #328: Whilst attempting to flash the Zigbee gateway onto the ESP32, it persistently reboots during execution.

yiwang103, 2024-08-21:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

esp32

### What is the expected behavior?

can running success.

### What is the actual behavior?

It keeps restarting.

### Steps to reproduce.

1. Step
2. Step
3. Step
...
in examples/zigbee_gateway use "idf.py build"
then flash to esp32
Power-on start
Then I checked the logs and found that it keeps restarting.

### Debug Logs.

```plain
[2024-08-21 10:20:33.860]
RX：[0;33mW(2934) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout[0m
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15648
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3904
entry 0x40080640
[0;32mI (31) boot: ESP-IDF v5.3 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Aug 21 2024 08:45:36[0m
[0;32mI (31) boot: Multicore bootloader[0m
[0;32mI (35) boot: chip revision: v3.1[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (44) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (53) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000[0m
[0;32mI (76) boot:  1 nvs              WiFi data        01 02 00010000 00006000[0m
[... 174 lines cut ...]
[0;32mI (786) app_init: Compile time:     Aug 21 2024 08:45:27[0m
[0;32mI (792) app_init: ELF file SHA256:  d56caf346...[0m
[0;32mI (797) app_init: ESP-IDF:          v5.3[0m
[0;32mI (802) efuse_init: Min chip rev:     v0.0[0m
[0;32mI (807) efuse_init: Max chip rev:     v3.99 [0m
[0;32mI (811) efuse_init: Chip rev:         v3.1[0m
[0;32mI (817) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (824) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (830) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (836) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (842) heap_init: At 3FFCF840 len 000107C0 (65 KiB): DRAM[0m
[0;32mI (848) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (854) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (861) heap_init: At 4009DB58 len 000024A8 (9 KiB): IRAM[0m
[0;32mI (868) spi_flash: detected chip: gd[0m
[0;32mI (871) spi_flash: flash io: dio[0m
[0;33mW (875) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m
[0;32mI (889) coexist: coex firmware version: dab85ae96[0m
[0;32mI (894) main_task: Started on CPU0[0m
[0;32mI (904) main_task: Calling app_main()[0m
[0;32mI (914) ESP_RADIO_SPINEL: spinel UART interface initialization completed[0m
[0;32mI (914) ESP_RADIO_SPINEL: Spinel UART interface has been successfully enabled[0m
[0;32mI (924) ZB_ESP_SPINEL: Spinel UART interface enable successfully[0m
[0;32mI (924) main_task: Returned from app_main()[0m
```

chshu, 2024-08-21:
The log indicates that the RCP is not well connected to the Host SoC, could you double check your hardware setup, and the RCP firmware is flashed correctly.

You may refer to the following links:
https://github.com/espressif/esp-rainmaker/tree/master/examples/zigbee_gateway#wi-fi-based-esp-zigbee-gateway
https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway#hardware-platforms

yiwang103, 2024-08-21:
> 日志表明 RCP 与主机 SoC 连接不良，您可以仔细检查硬件设置，并且 RCP 固件是否正确刷新。
> 
> 您可以参考以下链接： [https://github.com/espressif/esp-rainmaker/tree/master/examples/zigbee_gateway#wi-fi-based-esp-zigbee-gateway ](https://github.com/espressif/esp-rainmaker/tree/master/examples/zigbee_gateway#wi-fi-based-esp-zigbee-gateway) https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway#hardware-platforms

Is this RCP mandatory? In other words, do I need an additional ESP32-H besides the ESP32

chshu, 2024-08-21:
> Is this RCP mandatory? In other words, do I need an additional ESP32-H besides the ESP32

Yes, it's mandatory, since there is no 802.15.4 (zigbee) radio in ESP32. Zigbee interface is required to build a Zigbee Gateway.

[esp-zigbee-gateway-board](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway#esp-zigbee-gateway-board) is recommened to run the example, otherwise you need to connect two standalong SoCs manually.

yiwang103, 2024-08-21:
> > 这个 RCP 是强制性的吗？换句话说，除了 ESP32 之外，我还需要额外的 ESP32-H 吗？
> 
> 是的，这是强制性的，因为 ESP32 中没有 802.15.4 (zigbee) 无线电。构建 Zigbee 网关需要 Zigbee 接口。
> 
> [建议使用 esp-zigbee-gateway-board](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway#esp-zigbee-gateway-board)来运行示例，否则您需要手动连接两个独立的 SoC。

Okay, thank you for your response. I'll give it another try once I'm ready.

shahpiyushv, 2024-09-04:
Closing this since an answer has been provided.