espressif/esp-rainmaker #336: Header file missing

aadhimellow, 2024-10-07:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.2.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32S3 BOX 3

### What is the expected behavior?

The Matter controller with touchscreen + Rainmaker example with thread border router should build for the command `idf.py build` from repository https://github.com/espressif/esp-rainmaker/tree/master/examples/matter .

### What is the actual behavior?

The building gets stopped when it try to build the OTBR feature with an error 
esp-rainmaker/examples/matter/matter_controller_with_touchscreen/main/app_main.cpp:30:10: fatal error: esp_matter_thread_br_cluster.h: No such file or directory
   30 | #include <esp_matter_thread_br_cluster.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[2127/2167] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/app_matter_controller.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_matter_controller.cpp.obj 
BUG: component_requirements.py: cannot match original component filename for source component main.

This is due to the app_matter_controller.cpp.obj is missing in the components folder.
I have searched for the file in the esp repositories but didn't find any.

[image]

### Steps to reproduce.

1. Followed the steps mentioned in https://github.com/espressif/esp-rainmaker/tree/master/examples/matter for open thread border router
2. Executed `idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults;sdkconfig.defaults.otbr" set-target esp32s3`
3. Executed idf.py build and encountered with error at 2131/2167 of the build
...

### Build Logs.

```plain
/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/main/app_main.cpp:30:10: fatal error: esp_matter_thread_br_cluster.h: No such file or directory
   30 | #include <esp_matter_thread_br_cluster.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[2127/2167] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/app_matter_controller.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_matter_controller.cpp.obj 
/Users/a1248986/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.2.2\" -DLV_CONF_INCLUDE_SIMPLE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMD5_ENABLED=1 -DOPENTHREAD_CONFIG_FILE=\"openthread-core-esp32x-ftd-config.h\" -DOPENTHREAD_FTD=1 -DSERIAL_FLASHER_BOOT_HOLD_TIME_MS=50 -DSERIAL_FLASHER_RESET_HOLD_TIME_MS=100 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/build/config -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/main -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/main/box -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/main/box/gui -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/main/dynamic_qrcode -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/newlib/platform_include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/config/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/config/include/freertos -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/config/xtensa/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/freertos/esp_additions/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_hw_support/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_hw_support/include/soc -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_hw_support/include/soc/esp32s3 -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_hw_support/port/esp32s3/. -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/heap/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/log/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/soc/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/soc/esp32s3 -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/soc/esp32s3/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/hal/platform_port/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/hal/esp32s3/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/hal/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_rom/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_rom/include/esp32s3 -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_rom/esp32s3 -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_common/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_system/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_system/port/soc -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_system/port/include/private -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/xtensa/esp32s3/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/xtensa/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/xtensa/deprecated_include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_timer/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/include/apps -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/include/apps/sntp -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/lwip/src/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/port/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/port/freertos/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/port/esp32xx/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/port/esp32xx/include/arch -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/lwip/port/esp32xx/include/sys -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/managed_components/espressif__esp_secure_cert_mgr/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/spi_flash/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/port/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/mbedtls/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/mbedtls/library -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/esp_crt_bundle/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/mbedtls/3rdparty/everest/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/mbedtls/3rdparty/p256-m -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/nvs_flash/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/esp_partition/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/efuse/include -I/Users/a1248986/matterProject/esp-idf-v5.2.2/components/efuse/esp32s3/include -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/managed_components/espressif__led_strip/include -I/Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/managed_component [... cut ...]
[2131/2167] Building CXX object esp-idf/esp_matter_controller/CMakeFiles/__idf_esp_matter_controller.dir...ctedhomeip/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/logging/DataModelLogger.cpp.obj
ninja: build stopped: subcommand failed.
BUG: component_requirements.py: cannot match original component filename for source component main
ninja failed with exit code 1, output of the command is in the /Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/build/log/idf_py_stderr_output_46253 and /Users/a1248986/matterProject/esp-rainmaker/examples/matter/matter_controller_with_touchscreen/build/log/idf_py_stdout_output_46253
```

jsheo-at-konai, 2024-10-11:
same here.  Where can I find those files?

aadhimellow, 2024-10-15:
@shubhamdp could you please look into this issue ?

sayondeep, 2024-10-15:
@aadhimellow 
The issue has been identified and the fix is under review.
It will be merged on github soon.

Mr-Techtron, 2024-12-03:
@sayondeep any update on this?

chshu, 2025-01-10:
It should be fixed by https://github.com/espressif/esp-rainmaker/commit/97ee869a3220b9a392d3764ee24458ad41ba693a, please pull the latest esp-rainmaker and try again.