espressif/esp-rainmaker #337: Failed to Provision device in rainmaker

aadhimellow, 2024-10-15:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v5.3.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP Wi-Fi based Thread Border Router

### What is the expected behavior?

Tried to create a rainmaker matter controller with OTBR feature [on](https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/matter_controller) Espressif thread border router which should show the qr code and details for provisioning with rainmaker account and start thread network after that.

### What is the actual behavior?

After executing the command it shows an error failed to get device certificate. The qr code or provisioning details are not showing up which should be scanned on the rainmaker app.

### Steps to reproduce.

1. Cloned the repository `git clone --recursive https://github.com/espressif/esp-rainmaker.git`
2. Installed the prerequisites given in `https://github.com/espressif/esp-rainmaker/tree/master/examples/matter`
3. Created the environments
4. Followed the steps for claiming the certification
5. Generated the factory nvs binary
6. flashed it using esptool.py
7. Opened the example /examples/matter/matter_controller
8. Executed the steps mentioned and ran flash monitor
9. Got error failed to fetch device certificate along with other errors
[image]

### Debug Logs.

```plain
I (26) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (27) boot: compile time Oct 15 2024 16:59:34
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (28) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (29) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (30) boot:  2 otadata          OTA data         01 00 0001c000 00002000
I (30) boot:  3 phy_init         RF data          01 01 0001e000 00001000
I (30) boot:  4 ota_0            OTA app          00 10 00020000 002ee000
I (31) boot:  5 fctry            WiFi data        01 02 00320000 00006000
I (31) boot: End of partition table
I (31) esp_image: segment 0: paddr=00020020 vaddr=3c1f0020 size=86264h (549476) map
I (130) esp_image: segment 1: paddr=000a628c vaddr=3fc9e700 size=05b78h ( 23416) load
I (136) esp_image: segment 2: paddr=000abe0c vaddr=40374000 size=0420ch ( 16908) load
I (140) esp_image: segment 3: paddr=000b0020 vaddr=42000020 size=1e8b48h (2001736) map
I (499) esp_image: segment 4: paddr=00298b70 vaddr=4037820c size=164a0h ( 91296) load
I (519) esp_image: segment 5: paddr=002af018 vaddr=600fe100 size=00004h (     4) load
I (531) boot: Loaded app from partition at offset 0x20000
[... 114 lines cut ...]
I (1569) esp_claim: Private key already exists. No need to re-initialise it.
I (1579) esp_rmaker_node: Node ID ----- E80690D3FB84
I (1579) esp_rmaker_ota_using_topics: OTA enabled with Topics
I (1579) esp_rmaker_ota: OTA state = 2
I (1579) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (1579) esp_rmaker_time_service: Time service enabled
I (1579) esp_rmaker_time: SNTP already initialized.
I (1579) app_insights: Enable CONFIG_ESP_INSIGHTS_ENABLED to get Insights.
I (1579) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x131BFC00's Attribute 0x00000001 is E80690D3FB84 **********
I (1579) app_matter: RainMaker init not done. Not processing attribute update
I (1589) esp_rmaker_core: Starting RainMaker Work Queue task
I (1589) esp_rmaker_work_queue: RainMaker Work Queue task started.
W (1589) wifi:Haven't to connect to a suitable AP now!
I (1589) esp_claim: Waiting for assisted claim to finish.
I (1629) main_task: Returned from app_main()
>
```

### More Information.

While building the example a warning was shown but didn't cause any failure
`[1710/1714] Generating ld/sections.ld
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/a1248986/matterProject/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:966, /Users/a1248986/matterProject/esp-matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /Users/a1248986/matterProject/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:966, /Users/a1248986/matterProject/esp-matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
[1712/1714] Generating binary image from built executable`

shahpiyushv, 2024-10-15:
All Matter examples need to follow the instructions [here](https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md) to get the RainMaker MQTT Credentials/Matter DAC and other matter specific configuration. Have you done all of that?

One misleading instruction I see for TBR [here](https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/matter_controller#build) is that it also asks to erase flash, which will also clear the certificates. Instead, the instruction should have been just `idf.py -p <PORT> flash monitor`.

aadhimellow, 2024-10-15:
I have followed the instructions given in the link [``](https://github.com/espressif/esp-rainmaker/blob/master/examples/matter/README.md).
I have tried without giving erase-flash just like you suggested but result in an error

`E (1453) mdns: mdns_service_remove_for_host(6377): Invalid state or arguments
E (1453) mdns: mdns_service_remove_for_host(6377): Invalid state or arguments
E (1453) mdns: mdns_service_remove_for_host(6377): Invalid state or arguments
E (1453) chip[DIS]: Setup discriminator read error (201)! Critical error, will not be commissionable.
E (1453) chip[DIS]: Failed to advertise extended commissionable node: 201
E (1453) chip[SVR]: ERROR setting up transport: 201
I (1453) esp_matter_core: Dynamic endpoint 0 added
I (1453) esp_matter_core: Dynamic endpoint 1 added
I (1453) chip[DL]: WIFI_EVENT_STA_START
W (1453) wifi:Haven't to connect to a suitable AP now!
I (1453) chip[DL]: Done driving station state, nothing else to do...
I (1463) chip[DIS]: Updating services using commissioning mode 0
E (1463) mdns: mdns_service_remove_for_host(6377): Invalid state or arguments
E (1463) mdns: mdns_service_remove_for_host(6377): Invalid state or arguments
E (1463) mdns: mdns_service_remove_for_host(6377): Invalid state or arguments
E (1463) chip[DIS]: Setup discriminator read error (201)! Critical error, will not be commissionable.
E (1463) chip[DIS]: Failed to advertise extended commissionable node: 201
E (1463) esp_rmaker_core: Failed to initialise Node Id. Please perform "claiming" using RainMaker CLI.
E (1463) app_main: Could not initialise node.
`

sayondeep, 2024-10-15:
@aadhimellow please try by flashing the the generated factory partition at 0x320000 and reclaiming. 

I would suggest to erase-flash then, flash the firmware , then flash the factory partition and then claim. This is to avoid problems that can occur if you flashed the factory partition at incorrect address earlier.

aadhimellow, 2024-10-16:
@sayondeep I tried this approach and was able to create the controller successfully. Thank you for the support.
I were able to add two matter devices to the rainmaker app.