espressif/esp-rainmaker #338: Sending matter sensor data to rainmaker

aadhimellow, 2024-10-17:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I have flashed the matter controller + rainmaker example with otbr feature to an espressif thread border router and was able to successfully commission a matter door sensor and mattet plug to rainmaker. How can I send the states of those devices to the rainmaker dashboard or aws iot core. Currently one device is not showing any parameters which is the door lock sensor and the matter plug can be turned on and off with rainmaker but those two devices are not shown in rainmaker dashboard.
[image]
[image]

shahpiyushv, 2024-10-18:
The ESP RainMaker dashboard is an [admin side](https://rainmaker.espressif.com/docs/user-roles#admin-users) device management dashboard meant for devices directly connected to the RainMaker backend, which in this case, is the controller. The devices connected to the controller are pure matter devices, not relevant to admin users. They are accessible only to the [end user](https://rainmaker.espressif.com/docs/user-roles#end-users) via the phone app. Moreover, some device types are not yet supported, either by the controller, or the phone app. So, even if they may be controllable by API, they may not show correctly in the app.

aadhimellow, 2024-10-18:
Is there any possible way to send the data of the sensor to aws iot core via the matter controller

sayondeep, 2024-10-18:
You can check the supported device type by the controller here https://github.com/espressif/esp-rainmaker/blob/f98cf1ec50bff6706c5afe626806fe9d95dbc141/examples/matter/matter_controller_with_touchscreen/main/app_matter_ctrl.cpp#L399 

**Currently door lock sensor is not supported by the controller and phone app.**

Assuming that door lock sensor is supported in the controller. As a proof of concept you can your logic for sending on-off data of light to aws iot core by adding your logic here https://github.com/espressif/esp-rainmaker/blob/f98cf1ec50bff6706c5afe626806fe9d95dbc141/examples/matter/matter_controller_with_touchscreen/main/app_matter_ctrl.cpp#L120 .

We report the updated state of light (A supported device type by controller) .
To check and verify the state. You can run the following commands.
```
$ cd $RMAKER_PATH/
$ esp-rainmaker-cli login --email <rainmaker-account-email>
$ esp-rainmaker-cli  getparams <rainmaker-node-id-of-controller>
```
On-Off cluster is 0x06.

aadhimellow, 2024-10-21:
Thank you so much. Could you please tell me how to change the startup logo from espressif to a custom logo. I have seen a file in the location [matter controller with touchscreen](https://github.com/espressif/esp-rainmaker/tree/master/examples/matter/matter_controller_with_touchscreen/main/box/gui/image) in the file esp_logo.c. But I can see the image is converted into a hex format. How can I convert my custom logo like that. Also what should be the resolution and size of the logo?

sayondeep, 2024-10-21:
@aadhimellow For conversion of image please refer [here](https://lvgl.io/tools/imageconverter) use the lvgl v8 tool.

aadhimellow, 2024-10-23:
After commissioning the matter controller to rainmaker. I can see the current time in the s3box. How can I fetch that time and print it in the log.
After provisioning i have seen the date time details in the monitor under the esp log tag called esp_rmaker_time but couldnt find the function that does it

closed 2024-10-24 by aadhimellow