espressif/esp-rainmaker #343: error: void value not ignored as it ought to be

dzungpv, 2025-02-23:
### Answers checklist.

- [x] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [x] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [x] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32 DEV

### What is the expected behavior?

Build error on the MacOs

1.
```
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_node_config.c: In function '__esp_rmaker_get_node_config':
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_node_config.c:271:12: error: void value not ignored as it ought to be
  271 |     return json_gen_str_end(&jstr);
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_node_config.c:272:1: error: control reaches end of non-void function [-Werror=return-type]
  272 | }
      | ^
```

2, I try to fix 1 error show this error
```
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_param.c: In function 'esp_rmaker_populate_params':
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_param.c:165:14: error: void value not ignored as it ought to be
  165 |     *buf_len = json_gen_str_end(&jstr);
      |              ^

```

On the previous version of Mac it working fine, code is latest from this repo

### What is the actual behavior?

Build should work

### Steps to reproduce.

Build sample homekit_switch

### Build Logs.

```plain
1.

/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_node_config.c: In function '__esp_rmaker_get_node_config':
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_node_config.c:271:12: error: void value not ignored as it ought to be
  271 |     return json_gen_str_end(&jstr);
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_node_config.c:272:1: error: control reaches end of non-void function [-Werror=return-type]
  272 | }
      | ^

2, I try to fix 1 error show this error

/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_param.c: In function 'esp_rmaker_populate_params':
/Users/vietdzung/dev/projects/esp32/sdks/5.3.2/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_param.c:165:14: error: void value not ignored as it ought to be
  165 |     *buf_len = json_gen_str_end(&jstr);
      |              ^
```

vikramdattu, 2025-02-25:
@dzungpv the error means, the compiler thinks `json_gen_str_end` returns no value (i.e., `void`) but we are catching the return value. This is of course, wrong as the API returns the length of the json string.

Q: Did you modify the above said API's return value to be void?

dzungpv, 2025-02-25:
No, it is original code. Strange that it only show error with MacOSx 15.3.1 I have other Mac with 15.3.0 and not error with same code. May be gcc on 15.3.1 change. I try on Windows 11 it build ok too. I forgot to mention that I also using Arduino as IDF component with the sample.

vikramdattu, 2025-02-25:
Thanks for the confirmation! Strange behaviour indeed. If you are already using the IDF build system, using Arduino in the mix seems an odd choice. I am not sure, how and why you are putting it all together.

Few suggestions:
- Please do a grep for `json_gen_str_end` in all of your example sources and see if you hit something useful?
- Can you try your project with just the ESP-IDF (no Arduino) or would that be too much of a change?

dzungpv, 2025-02-26:
> Thanks for the confirmation! Strange behaviour indeed. If you are already using the IDF build system, using Arduino in the mix seems an odd choice. I am not sure, how and why you are putting it all [...]

This is output of : `grep -r -n "json_gen_str_end" .` command
```
% grep -r -n "json_gen_str_end" .
./build/log/idf_py_stdout_output_4937:1791:  165 |     *buf_len = json_gen_str_end(&jstr);
./build/log/idf_py_stdout_output_4937:1798:  271 |     return json_gen_str_end(&jstr);
./build/log/idf_py_stdout_output_4937:1816:  901 |     *buf_size = json_gen_str_end(&jstr);
Binary file ./build/esp-idf/esp_hap_core/CMakeFiles/__idf_esp_hap_core.dir/src/esp_hap_ip_services.c.obj matches
Binary file ./build/esp-idf/esp_rainmaker/CMakeFiles/__idf_esp_rainmaker.dir/src/core/esp_rmaker_user_mapping.c.obj matches
Binary file ./build/esp-idf/json_generator/libjson_generator.a matches
Binary file ./build/esp-idf/json_generator/CMakeFiles/__idf_json_generator.dir/upstream/json_generator.c.obj matches
```