espressif/esp-rainmaker #349: Confirming Node association failed when trying switch example

iBacklight, 2025-03-24:
### Answers checklist.

- [x] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [x] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [x] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I have successfully build the project of the switch example using idf.py. I also erased flash using idf.py erase_flash command. After flash the code into the ESP32-C3 dev board and use rainmaker to scan the QR code to add device, it has stuck in the section of `Confirming Node association` while provisioning. I didn't change anything within the project.

Here is the full log after I scan the QR code:
I (20618) protocomm_nimble: mtu update event; conn_handle=1 cid=4 mtu=256
I (21488) app_network: Secured session established!
I (28808) esp_rmaker_user_mapping: Received request for node details
I (28808) esp_rmaker_user_mapping: Got user_id = 465a806d-a2b2-4704-a678-237b4d17ae86, secret_key = 1320F13E-9EC9-4C5E-9DC9-45F00862A244
I (28818) esp_rmaker_user_mapping: User Node mapping reset detected.
I (28828) esp_rmaker_user_mapping: Sending status SUCCESS
W (29078) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (29098) app_wifi: Received Wi-Fi credentials
        SSID     : XXXXXXXXX
        Password : XXXXXXXXXX
I (35198) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (35198) wifi:state: init -> auth (0xb0)
I (35208) wifi:state: auth -> assoc (0x0)
I (35208) wifi:state: assoc -> run (0x10)
I (35238) wifi:connected with TELUS5241, aid = 9, channel 1, BW20, bssid = 74:90:bc:55:a4:fd
I (35238) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (35258) wifi:pm start, type: 1

I (35258) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (35258) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (35328) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (36868) wifi:<ba-add>idx:0 (ifx:0, 74:90:bc:55:a4:fd), tid:0, ssn:1, winSize:64
I (37768) esp_netif_handlers: sta ip: 192.168.1.73, mask: 255.255.255.0, gw: 192.168.1.254
I (37768) app_network: Connected with IP Address:192.168.1.73
I (37768) network_prov_mgr: STA Got IP
I (37768) app_wifi: Provisioning successful
I (37778) app_main: RainMaker Claim Started.
I (37778) esp_claim: Starting the Self Claim Process. This may take time.
I (37798) main_task: Returned from app_main()
I (38108) esp-x509-crt-bundle: Certificate validated
I (40038) esp-x509-crt-bundle: Certificate validated
I (40368) esp_rmaker_time: SNTP Synchronised.
I (40368) esp_rmaker_time: The current time is: Mon Mar 24 08:24:13 2025 +0800[CST], DST: No.
I (40458) NimBLE: GAP procedure initiated: stop advertising.

I (40458) NimBLE: GAP procedure initiated: stop advertising.

I (40468) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

E (40508) protocomm_nimble: Error setting advertisement data; rc = 30
I (40518) network_prov_mgr: Provisioning stopped
I (40518) network_prov_scheme_ble: BTDM memory released
I (40518) esp_rmaker_local: Event 6
I (41798) esp_claim: Self Claiming was successful. Certificate received.
I (41798) app_main: RainMaker Claim Successful.
I (41808) esp_mqtt_glue: AWS PPI: ?Platform=APN3|A0|RM|EX00|RMDev|1x0|5432044B9D88
I (41808) esp_mqtt_glue: Initialising MQTT
I (41818) esp_rmaker_mqtt_budget: MQTT Budgeting initialised. Default: 100, Max: 1024, Revive count: 1, Revive period: 5
I (41828) esp_rmaker_cmd_resp: Enabling Command-Response Module.
I (41838) esp_rmaker_node: Node attribute cmd-resp created
I (41838) esp_rmaker_local: Enabling Local Control
I (41848) esp_rmaker_local: Couldn't find POP in NVS. Generating a new one.
I (41858) esp_rmaker_local: Local Control Service Enabled
I (41858) esp_rmaker_local: Starting ESP Local control with HTTP Transport and security version: 1
I (41868) mdns_mem: mDNS task will be created from internal RAM
I (41878) esp_https_server: Starting server
I (41888) esp_https_server: Server listening on port 8080
I (41888) esp_rmaker_local: esp_local_ctrl service started with name : 5432044B9D88
I (41898) app_main: Local Control Started.
I (41898) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (41908) esp_rmaker_core: Waiting for MQTT connection
I (42508) esp-x509-crt-bundle: Certificate validated
I (43798) esp_mqtt_glue: MQTT Connected
I (43798) app_main: MQTT Connected.
I (43798) esp_rmaker_cmd_resp: Checking for pending commands.
I (43808) esp_rmaker_node_config: Generated Node config of length 1458
I (43808) esp_rmaker_core: Waiting for User Node Association.
I (43818) esp_rmaker_ota_using_topics: Subscribing to: node/5432044B9D88/otaurl
I (43828) esp_rmaker_user_mapping: MQTT Publish: {"node_id":"5432044B9D88","user_id":"465a806d-a2b2-4704-a678-237b4d17ae86","secret_key":"1320F13E-9EC9-4C5E-9DC9-45F00862A244","reset":true}
I (43888) app_main: MQTT Published. Msg id: 22582.
I (43958) app_main: MQTT Published. Msg id: 35038.
I (43968) app_main: MQTT Published. Msg id: 60137.
I (43968) esp_rmaker_user_mapping: User Node association message published successfully.
W (43968) app_main: Unhandled RainMaker Event: 5
I (43978) esp_rmaker_param: Params MQTT Init done.
I (43978) esp_rmaker_param: Reporting params (init): {"Switch":{"Name":"Switch","Power":true},"Time":{"TZ":"","TZ-POSIX":""},"Schedule":{"Schedules":[]},"Scenes":{"Scenes":[]},"Local Control":{"POP":"d506ff5f","Type":1}}
I (44468) app_main: MQTT Published. Msg id: 55690.
I (48828) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (49058) app_main: MQTT Published. Msg id: 47090.
I (50898) wifi:<ba-add>idx:1 (ifx:0, 74:90:bc:55:a4:fd), tid:3, ssn:1, winSize:64

It is noticed that there is only an error: 
`E (40508) protocomm_nimble: Error setting advertisement data; rc = 30`
and several warnings:
`W (29078) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2`
`W (43968) app_main: Unhandled RainMaker Event: 5`

Thank you for your help in advance!

shahpiyushv, 2025-03-24:
The node side logs themselves look fine. Is the phone app directly downloaded from the app store? Is it iOS or Android? Can you share screenshot of the phone app screen?

iBacklight, 2025-03-25:
Yes, I used an iPhone and downloaded the app directly from the App Store. The screenshot was like:

[image]

I have tried to reset the Wi-Fi and factory settings using the boot button, but the problem still exists. BTW I'm using esp idf 5.4.

Thank you!

shahpiyushv, 2025-03-26:
Our backend team checked the logs for this node and informed that the user id passed by the node doesn't exist. This typically happens when the phone app is configured to work with a private RainMaker deployment. Is it what you have been trying?

iBacklight, 2025-03-27:
Hello,

Yes probably I have used apple id to login in a private way. Also the app seems to be switched to another region, which caused a confirming error. After I chose to share my apple id with rainmaker and switch back to my current region, the switch demo just worked.

Thank you again for your assistance!

closed 2025-03-27 by iBacklight