espressif/esp-rainmaker #353: switch example fails to compile

14790897, 2025-05-22:
### Answers checklist.

- [x] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [x] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [x] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

v5.5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### Development Kit.

ESP32-C3

### What is the expected behavior?

build successfully

### What is the actual behavior?

Merged 2 ELF sections
Successfully created esp32c3 image.
Generated C:/git-program/esp-rainmaker/examples/switch/build/bootloader/bootloader.bin
[109/109] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\git-program\esp-rainmaker\examples\switch\build\boo...et 0xc000 bootloader 0x0 C:/git-program/esp-rainmaker/examples/switch/build/bootloader/bootloader.bin 
Bootloader binary size 0x52d0 bytes. 0x6d30 bytes (57%) free.
[1238/1240] Linking CXX executable switch.elf
FAILED: switch.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Users\13963\.espressif\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=C:/git-program/esp-rainmaker/examples/switch/build/switch.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.bt_funcs.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T esp32c3.rom.eco3_bt_funcs.ld -T esp32c3.rom.libc.ld -T esp32c3.rom.libc-suboptimal_for_misaligned_mem.ld -T esp32c3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\switch.elf.rsp -o switch.elf && cd ."
C:/Users/13963/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj): in function esp_log_writev_wrapper':
C:/git-program/esp-idf/components/esp_wifi/esp32c3/esp_adapter.c:358:(.text.esp_log_writev_wrapper+0x4): undefined reference to __wrap_esp_log_writev'
C:/Users/13963/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj): in function esp_log_write_wrapper':
C:/git-program/esp-idf/components/esp_wifi/esp32c3/esp_adapter.c:365:(.text.esp_log_write_wrapper+0x12): undefined reference to __wrap_esp_log_writev'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\git-program\esp-rainmaker\examples\switch\build\log\idf_py_stderr_output_51532 and C:\git-program\esp-rainmaker\examples\switch\build\log\idf_py_stdout_output_51532

### Steps to reproduce.

1. idf.py set-target esp32c3
2. idf.py build 

### Build Logs.

```plain

```

14790897, 2025-05-22:
```
PS C:\git-program\esp-rainmaker\examples\switch> idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\git-program\esp-rainmaker\examples\switch\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory C:\git-program\esp-rainmaker\examples\switch\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\13963\.espressif\python_env\idf5.5_py3.12_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 C:\git-program\esp-rainmaker\examples\switch"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.41.0.windows.1")
-- Minimal build - OFF
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/13963/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/13963/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/13963/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
NOTICE: Using component placed at C:\git-program\esp-rainmaker\components\esp_rainmaker for dependency "espressif/esp_rainmaker", specified in C:\git-program\esp-rainmaker\examples\switch\main\idf_component.yml
NOTICE: Using component placed at C:\git-program\esp-rainmaker\components\esp_rainmaker for dependency "espressif/esp_rainmaker", specified in C:\git-program\esp-rainmaker\examples\common\app_insights\idf_component.yml
NOTICE: 
Following dependencies have new versions available:
Dependency "espressif/cbor": "0.6.0~1" -> "0.6.0~2"
Consider running "idf.py update-dependencies" to update your lock file.
NOTICE: Processing 17 dependencies:
NOTICE: [1/17] espressif/cbor (0.6.0~1)
NOTICE: [2/17] espressif/esp-serial-flasher (0.0.11)
NOTICE: [3/17] espressif/esp_diag_data_store (1.0.2)
NOTICE: [4/17] espressif/esp_diagnostics (1.2.1)
NOTICE: [5/17] espressif/esp_insights (1.2.4)
NOTICE: [6/17] espressif/esp_rainmaker (1.5.4) (C:\git-program\esp-rainmaker\components\esp_rainmaker)
NOTICE: [7/17] espressif/esp_rcp_update (1.2.0)
NOTICE: [8/17] espressif/esp_schedule (1.2.0)
NOTICE: [9/17] espressif/esp_secure_cert_mgr (2.5.0)
NOTICE: [10/17] espressif/jsmn (1.1.0)
NOTICE: [11/17] espressif/json_generator (1.1.2)
NOTICE: [12/17] espressif/json_parser (1.0.3)
NOTICE: [13/17] espressif/mdns (1.8.2)
NOTICE: [14/17] espressif/network_provisioning (1.0.5)
NOTICE: [15/17] espressif/qrcode (0.1.0~2)
NOTICE: [16/17] espressif/rmaker_common (1.4.6)
NOTICE: [17/17] idf (5.5.0)
-- Project sdkconfig file C:/git-program/esp-rainmaker/examples/switch/sdkconfig
Loading defaults file C:/git-program/esp-rainmaker/examples/switch/sdkconfig.defaults...
C:/git-program/esp-rainmaker/examples/switch/sdkconfig.defaults:33 CONFIG_ESP_RMAKER_LOCAL_CTRL_ENABLE was replaced with CONFIG_ESP_RMAKER_LOCAL_CTRL_AUTO_ENABLE 
warning: unknown kconfig symbol 'MBEDTLS_DYNAMIC_FREE_PEER_CERT' assigned to 'y' in C:/git-program/esp-rainmaker/examples/switch/sdkconfig.defaults
warning: unknown kconfig symbol 'BTDM_CTRL_MODE_BLE_ONLY' assigned to 'y' in C:/git-program/esp-rainmaker/examples/switch/sdkconfig.defaults
-- Adding linker script C:/git-program/esp-idf/components/riscv/ld/rom.api.ld
-- Found Python3: C:/Users/13963/.espressif/python_env/idf5.5_py3.12_env/Scripts/python.exe (found version "3.12.7") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- USING O3
-- App "switch" version: 1.0
-- Adding linker script C:/git-program/esp-rainmaker/examples/switch/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/git-program/esp-rainmaker/examples/switch/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script C:/git-program/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script C:/git-program/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script C:/git-program/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.bt_funcs.ld
[... 80 lines cut ...]
-- Adding linker script C:/git-program/esp-idf/components/bootloader/subproject/main/ld/esp32c3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_security esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: C:/git-program/esp-idf/components/bootloader C:/git-program/esp-idf/components/bootloader_support C:/git-program/esp-idf/components/efuse C:/git-program/esp-idf/components/esp_app_format C:/git-program [... cut ...]  `esp_log_write_wrapper':
C:/git-program/esp-idf/components/esp_wifi/esp32c3/esp_adapter.c:365:(.text.esp_log_write_wrapper+0x12): undefined reference to `__wrap_esp_log_writev'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\git-program\esp-rainmaker\examples\switch\build\log\idf_py_stderr_output_34560 and C:\git-program\esp-rainmaker\examples\switch\build\log\idf_py_stdout_output_34560
```

vikramdattu, 2025-05-22:
@14790897 this is a known bug coming from ESP-Insights and we are in the process of fixing this.
As a workaround, since you are not using ESP-Insights, please disable `CONFIG_DIAG_ENABLE_WRAP_LOG_FUNCTIONS` from using `idf.py menuconfig`. The error should go away.

14790897, 2025-05-22:
@vikramdattu 

[image]

can't find this option

14790897, 2025-05-22:
[image]

vikramdattu, 2025-05-22:
Okay, here is a no-elegant solution:

Find: `esp_diagnostics/CMakeLists.txt` and comment out below contents:

```
# If log level is set to NONE or if logging APIs are externally wrapped then skip
# wrapping logging APIs here
if ((NOT CONFIG_LOG_DEFAULT_LEVEL_NONE) AND (NOT CONFIG_DIAG_USE_EXTERNAL_LOG_WRAP))
    list(APPEND WRAP_FUNCTIONS esp_log_write esp_log_writev)
endif()

if(CONFIG_LIB_BUILDER_COMPILE)
    list(APPEND WRAP_FUNCTIONS log_printf)
endif()

foreach(func ${WRAP_FUNCTIONS})
     target_link_libraries(${COMPONENT_LIB} INTERFACE "-Wl,--wrap=${func}")
endforeach()

```

closed 2025-05-22 by 14790897