espressif/esp-rainmaker #355: switch example fails to connect

14790897, 2025-05-22:
### Answers checklist.

- [x] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [x] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [x] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

```
I (54060) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (54060) wifi:state: init -> auth (0xb0)
I (54130) wifi:state: auth -> init (0xe00)
I (54140) wifi:Coexist: Wi-Fi connect fail, apply reconnect coex policy

I (54140) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (54140) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (54150) wifi:state: init -> auth (0xb0)
I (54170) wifi:state: auth -> assoc (0x0)
I (54170) wifi:state: assoc -> run (0x10)
I (54250) wifi:connected with HOME-GUEST, aid = 1, channel 1, 40U, bssid = aa:1a:3a:6d:ea:82
I (54250) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (54270) wifi:pm start, type: 1

I (54270) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (54270) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (54280) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (54620) wifi:<ba-add>idx:0 (ifx:0, aa:1a:3a:6d:ea:82), tid:0, ssn:0, winSize:64
I (55760) esp_netif_handlers: sta ip: 192.168.0.114, mask: 255.255.255.0, gw: 192.168.0.1
I (55760) app_network: Connected with IP Address:192.168.0.114
I (55760) network_prov_mgr: STA Got IP
I (55760) app_wifi: Provisioning successful
I (55770) app_main: RainMaker Claim Started.
I (55770) esp_claim: Starting the Self Claim Process. This may take time.
I (55780) main_task: Returned from app_main()
I (57040) esp-x509-crt-bundle: Certificate validated
I (60520) esp-x509-crt-bundle: Certificate validated
I (61790) NimBLE: GAP procedure initiated: stop advertising.

I (61790) NimBLE: GAP procedure initiated: stop advertising.

I (61790) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

E (61820) protocomm_nimble: Error setting advertisement data; rc = 30
I (61830) network_prov_mgr: Provisioning stopped
I (61830) network_prov_scheme_ble: BTDM memory released
I (61830) esp_rmaker_local: Event 6
I (63910) esp_claim: Self Claiming was successful. Certificate received.
I (63920) app_main: RainMaker Claim Successful.
I (63920) esp_mqtt_glue: AWS PPI: ?Platform=APN3|A0|RM|EX00|RMDev|1x0|34CDB0A7CA94
I (63920) esp_mqtt_glue: Initialising MQTT
I (63930) esp_rmaker_mqtt_budget: MQTT Budgeting initialised. Default: 100, Max: 1024, Revive count: 1, Revive period: 5
I (63940) esp_rmaker_cmd_resp: Enabling Command-Response Module.
I (63940) esp_rmaker_node: Node attribute cmd-resp created
I (63950) esp_rmaker_local: Enabling Local Control
I (63950) esp_rmaker_local: Couldn't find POP in NVS. Generating a new one.
I (63960) esp_rmaker_local: Local Control Service Enabled
I (63960) esp_rmaker_local: Starting ESP Local control with HTTP Transport and security version: 1
I (63970) mdns_mem: mDNS task will be created from internal RAM
I (63980) esp_https_server: Starting server
I (63980) esp_https_server: Server listening on port 8080
I (63990) esp_rmaker_local: esp_local_ctrl service started with name : 34CDB0A7CA94
I (63990) app_main: Local Control Started.
I (64000) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (64010) esp_rmaker_core: Waiting for MQTT connection
I (64990) esp-x509-crt-bundle: Certificate validated
I (67890) esp_mqtt_glue: MQTT Connected
I (67890) app_main: MQTT Connected.
I (67890) esp_rmaker_cmd_resp: Checking for pending commands.
I (67900) esp_rmaker_node_config: Generated Node config of length 1458
I (67900) esp_rmaker_core: Waiting for User Node Association.
I (67910) esp_rmaker_ota_using_topics: Subscribing to: node/34CDB0A7CA94/otaurl
I (67920) esp_rmaker_user_mapping: MQTT Publish: {"node_id":"34CDB0A7CA94","user_id":"db349089-284a-4743-90f8-e6f64e8ff0bc","secret_key":"313fd78c-67c5-44ab-8974-ef1aabfad6de","reset":true}
I (68410) app_main: MQTT Published. Msg id: 33891.
I (68810) app_main: MQTT Published. Msg id: 39196.
I (68810) app_main: MQTT Published. Msg id: 38221.
I (68810) esp_rmaker_user_mapping: User Node association message published successfully.
W (68820) app_main: Unhandled RainMaker Event: 5
I (68820) esp_rmaker_param: Params MQTT Init done.
I (68830) esp_rmaker_param: Reporting params (init): {"Switch":{"Name":"Switch","Power":true},"Time":{"TZ":"","TZ-POSIX":""},"Schedule":{"Schedules":[]},"Scenes":{"Scenes":[]},"Local Control":{"POP":"463d8aa9","Type":1}}
I (69940) app_main: MQTT Published. Msg id: 64362.
I (72910) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (73340) app_main: MQTT Published. Msg id: 47333.
I (83150) esp_rmaker_time: SNTP Synchronised.
I (83150) esp_rmaker_time: The current time is: Thu May 22 18:57:47 2025 +0800[CST], DST: No.
```

[image]

14790897, 2025-05-25:
help

shahpiyushv, 2025-06-03:
If you are using the Chinese variants of the ESP RainMaker phone app, you will have to enable Assisted Claiming.

idf.py menuconfig
  → ESP RainMaker Configuration
    → Claiming Type → Assisted Claiming

closed 2025-07-15 by 14790897